Gepetto/gepetto-viewer #32: devel -> master & release

olivier-stasse, 2016-05-06:
Dear all,
Will you be ok with merging devel into master,
and thinking about a release (the tag for v1.0.0 is 03/2015...)

Best,
Olivier.

florent-lamiraux, 2016-05-06:
We need to check that the devel branch is compatible with all other master branches.
I will do it.

jmirabel, 2016-05-06:
I agree with you @florent-lamiraux, though it is very unlikely that there is any problem. I merged master and devel branches not so long ago and there are very few changes in the viewer.

olivier-stasse, 2016-05-06:
Do you think it would be worthwhile to write a set of unitary tests to check if the API is broken ?

olivier-stasse, 2016-05-06:
This would avoid testing over all master branches...

jmirabel, 2016-05-06:
Adding tests would be useful but I'm not sure how easy it is to automate tests on a viewer... I'm not sure what the best approach is. Maybe it would be easier to do this in the `gepetto-viewer-corba` package.

olivier-stasse, 2016-05-06:
@jmirabl In this case it would consists simply on calling each method for each class.
Modification of the API woud automatically break the tests compilation ...

jmirabel, 2016-05-06:
Oh yes, I did not understand it this way.
It should help to keep backward compatibility.

florent-lamiraux, 2016-05-07:
I pushed release 1.1.0 on trac.laas.fr
I did the same for gepetto-viewer-corba.

closed 2016-05-07 by florent-lamiraux

olivier-stasse, 2016-05-07:
Dear Florent,
Thanks a lot.
Could you push the tags on the github repository ?