Gepetto/gepetto-viewer #57: Error compiling Gepetto-viewer on macOS 10.13 High Sierra

zongmianli, 2017-11-24:
I am trying to install the master branch on Mac OS 10.13. I encountered 3 types of errors while compiling urdf-parser.cpp, and each occurred multiple times.

Since I don't use the URDF parser in my project, is it easy to just remove that functionality while installing? Thanks in advance.

******1. no viable overloaded******

/Users/zml/gepetto-viewer/src/urdf-parser.cpp:105:18: error: no viable overloaded '='
          visual = link->visual_array[index];
          ~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:524:18: note: candidate function not viable: no known conversion from 'value_type' (aka 'std::__1::shared_ptr<urdf::Visual>') to 'const boost::shared_ptr<urdf::Visual>' for 1st argument
    shared_ptr & operator=( shared_ptr const & r ) BOOST_NOEXCEPT
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:533:18: note: candidate template ignored: could not match 'boost::shared_ptr' against 'std::__1::shared_ptr'
    shared_ptr & operator=(shared_ptr<Y> const & r) BOOST_NOEXCEPT
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:544:18: note: candidate template ignored: could not match 'auto_ptr' against 'shared_ptr'
    shared_ptr & operator=( std::auto_ptr<Y> & r )
                 ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:562:77: note: candidate template ignored: substitution failure [with Ap = std::__1::shared_ptr<urdf::Visual>]: no type named 'type' in 'boost::detail::sp_enable_if_auto_ptr<std::__1::shared_ptr<urdf::Visual>,
      boost::shared_ptr<urdf::Visual> &>'
    typename boost::detail::sp_enable_if_auto_ptr< Ap, shared_ptr & >::type operator=( Ap r )
                                                                       ~~~~ ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:584:18: note: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
    shared_ptr & operator=( boost::movelib::unique_ptr<Y, D> r )  

******2. no matching function for call to 'static_pointer_cast'******
/Users/zml/gepetto-viewer/src/urdf-parser.cpp:147:20: error: no matching function for call to 'static_pointer_cast'
        mesh_shared_ptr = boost::static_pointer_cast< ::urdf::Mesh >
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:845:42: note: candidate template ignored: could not match 'boost::shared_ptr' against 'std::__1::shared_ptr'
template<class T, class U> shared_ptr<T> static_pointer_cast( shared_ptr<U> const & r ) BOOST_NOEXCEPT

******3. use of overloaded operator '!=' is ambiguous******
/Users/zml/gepetto-viewer/src/urdf-parser.cpp:172:62: error: use of overloaded operator '!=' is ambiguous (with operand types 'MaterialSharedPtr' (aka 'shared_ptr<urdf::Material>') and 'long')
          if (visual && urdfLink->visual_array [j]->material != NULL) {
                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~

jmirabel, 2017-11-24:
The problem is that you have a more recent version of urdfdom, which uses `shared_ptr` from `std` instead of from `boost`. It should not be hard to solve. You could just try to replace the `boost::shared_ptr` by `std::shared_ptr`. (Same thing for cast operators). I do not guaranty it will work, but it is worth trying.

@nim65s Would you have time to make it compliant for both versions (as done in pinocchio) ?

jcarpent, 2017-11-24:
Or simply, you can install an older version of URDF (eg 0.3.0) waiting for the fix from our side.

jcarpent, 2017-11-24:
It appears that @nim65s already solved the issue (see https://github.com/nim65s/gepetto-viewer). It must be sufficient for your issues.

zongmianli, 2017-11-24:
Thanks Joseph and Justin, for being so reactive.

The `shared_ptr` problem has been solved by following Joseph's instruction (specifically I have modified 
> include/gepetto/viewer/macros.h
> include/gepetto/viewer/node-property.h
> src/urdf-parser.cpp

by making the same changes as commit 90e154 in the forked project https://github.com/nim65s/gepetto-viewer ). But the third error in the issue still exists: **use of overloaded operator '!=' is ambiguous**:

> /Users/zml/gepetto-viewer/src/urdf-parser.cpp:154:28: error: use of overloaded operator '!=' is ambiguous (with operand types '::std::shared_ptr< ::urdf::Mesh>' and 'int')
>       if ( mesh_shared_ptr != 0 )
>            ~~~~~~~~~~~~~~~ ^  ~
> /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:4961:1: note: candidate function [with _Tp = urdf::Mesh]
> operator!=(const shared_ptr<_Tp>& __x, nullptr_t) _NOEXCEPT
> ^
> /Users/zml/gepetto-viewer/src/urdf-parser.cpp:154:28: note: built-in candidate operator!=(int, int)
>       if ( mesh_shared_ptr != 0 )
>                            ^

When I chose to compile the forked project I got another error:

> /Users/zml/gepetto-viewer/src/window-manager.cpp:542:12: error: no member named 'get' in 'osg::Matrixd'
>     matrix.get(rot);
>     ~~~~~~ ^

jmirabel, 2017-11-25:
Replace `if ( mesh_shared_ptr != 0 )` by `if ( mesh_shared_ptr )` for the first error.

The second error has been fixed in the devel branch by #56. It will be integrated soon in the master branch. Either try the devel branch or use `git cherry-pick` to take the commit which fixed the issue.

closed 2017-11-26 by zongmianli