Gepetto/gepetto-viewer #66: issue with running CMake (maybe ADD_REQUIRED_DEPENDENCY macro)

miggia, 2018-06-01:
Hi,
I've been trying to compile gepetto-viewer on a Ubuntu distribution.
When I run `cmake`, however, I get the following error message.

```
-- openscenegraph >= 3.2 is required.
-- Checking for module 'openscenegraph >= 3.2'
--
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  cmake/pkg-config.cmake:288 (PKG_CHECK_MODULES)
  cmake/pkg-config.cmake:459 (ADD_DEPENDENCY)
  CMakeLists.txt:75 (ADD_REQUIRED_DEPENDENCY)
```

The issue seems to be related to the `ADD_REQUIRED_DEPENDENCY` macro in `cmake/pkg-config.cmake` failing to find `openscenegraph`.
`openscenegraph` is installed in `/usr/local/lib64/`; I tried both compiling and installing with `apt-get install` and it does not seem to make a difference.
I'm not used to the macro system being used for adding libraries to this project.
Any suggestion on how to fix/troubleshoot the issue?

jmirabel, 2018-06-01:
`/usr/local/lib64/pkgconfig`  must be in the `PKG_CONFIG_PATH` environment variable.

miggia, 2018-06-01:
That solved my issue; thank you.

Just for the record, when making the project I was getting `error: ‘to_string’ is not a member of ‘std’` errors, which I managed to solve by switching from `gcc5` to `gcc6`.

closed 2018-06-01 by miggia