Gepetto/gepetto-viewer #78: basic geometry from the urdf no recognised

MaximilienNaveau, 2019-02-11:
Upon loading of a urdf, the simple geometry are not parse properly. The viewer creates the nodes but do not display anything. I think this has to do with the urdfdom compatilibity.

jmirabel, 2019-02-11:
Hi,
Check the standard output for error messages.

MaximilienNaveau, 2019-02-14:
There are no output messages... That is why I need to go in debug mode and check things out...
Is this functionality working for you?
You can reproduce the behavior with this kind of link in a urdf:

```
  <link name="L_SHOULDER_HR">
    <visual>
      <geometry>
        <cylinder length="0.29845" radius=".02"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 -0.149225"/>
      <material name="dark_grey">
        <color rgba=".3 .3 .3 1.0"/>
      </material>
    </visual>
    <inertial>
      <mass value="2.865797"/>
      <origin rpy="0 0 0" xyz="0.002134 -0.004369 -0.185471"/>
      <inertia ixx="0.017193167054" ixy="0.000430328556" ixz="-3.4459702e-05"
                   yy="0.014573946765" iyz="-0.001182582441" izz="0.005103460355"/>
    </inertial>
  </link>
```

jmirabel, 2019-02-19:
There is a tiny error in your URDF. `yy` should be replaced by `iyy`.

The following works fine on my computer:
```py
from gepetto.corbaserver import Client
cl = Client()

try:
    cl.gui.getWindowID ("test")
except:
    cl.gui.createWindow ("test")
    # # You may need to call the two functions below if your version is old.
    # createGroup
    # addSceneToWindow

cl.gui.deleteNode ("foo", True)
cl.gui.addURDF ("foo", """
  <robot name="foo">
  <link name="L_SHOULDER_HR">
    <visual>
      <geometry>
        <cylinder length="0.29845" radius=".02"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 -0.149225"/>
      <material name="dark_grey">
        <color rgba=".3 .3 .3 1.0"/>
      </material>
    </visual>
    <inertial>
      <mass value="2.865797"/>
      <origin rpy="0 0 0" xyz="0.002134 -0.004369 -0.185471"/>
      <inertia ixx="0.017193167054" ixy="0.000430328556" ixz="-3.4459702e-05"
                   iyy="0.014573946765" iyz="-0.001182582441" izz="0.005103460355"/>
    </inertial>
  </link>
  </robot>
  """, "")
cl.gui.addToGroup ("foo", "test")
```

jmirabel, 2019-02-19:
I let you reopen this if necessary.

closed 2019-02-19 by jmirabel