Gepetto/gepetto-viewer #79: Scaling factor set in the urdf are not recognized.

MaximilienNaveau, 2019-02-14:
Setup:
Ubuntu14.04
robotpkg standard+wip with binary installation

example of problems with urdf:
```
<link name="L_HAND">
    <visual>
      <!-- <geometry>
          <box size=".06 .06 .06"/>
        </geometry> -->
      <geometry>
        <mesh filename="package://robot_properties_athena/meshes/stl/athena_hand.stl"
                    scale="0.001 0.001 0.001"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <material name="grey">
        <color rgba=".7 .7 .7 1.0"/>
      </material>
    </visual>
  </link>
```

jmirabel, 2019-02-14:
This is used by many people and I never had reports of such an issue. I
will have a look next week.

jmirabel, 2019-02-19:
The following code works fine on my computer:
```py
from gepetto.corbaserver import Client
cl = Client()

try:
    cl.gui.getWindowID ("test")
except:
    cl.gui.createWindow ("test")
    # # You may need to call the two functions below if your version is old.
    # createGroup
    # addSceneToWindow

cl.gui.deleteNode ("foo", True)
cl.gui.addURDF ("foo", """
  <robot name="foo">
<link name="L_HAND">
    <visual>
      <geometry>
        <mesh filename="package://ur_description/meshes/ur3/collision/base.stl"
                    scale="0.1 0.1 0.1"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 0.1"/>
      <material name="grey">
        <color rgba=".7 .7 .7 1.0"/>
      </material>
    </visual>
    <visual>
      <geometry>
        <mesh filename="package://ur_description/meshes/ur3/collision/base.stl"
                    scale="1 1 1"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <material name="red">
        <color rgba=".7 .1 .1 1.0"/>
      </material>
    </visual>
  </link>
  </robot>
  """, "")
cl.gui.addToGroup ("foo", "test")
```

jmirabel, 2019-02-19:
I let you reopen this if necessary.

closed 2019-02-19 by jmirabel