Gepetto/gepetto-viewer #80: error: ‘ObjectCache’ is not a member of ‘osgDB’

nim65s, 2019-02-28:
Hi,

Since #77, I can't build devel:
```
/local/users/gsaurel/gepetto/gepetto-viewer/src/leaf-node-collada.cpp: In member function ‘void graphics::LeafNodeCollada::init()’:
/local/users/gsaurel/gepetto/gepetto-viewer/src/leaf-node-collada.cpp:59:27: error: ‘ObjectCache’ is not a member of ‘osgDB’
       static osg::ref_ptr<osgDB::ObjectCache> object_cache (new osgDB::ObjectCache);
                           ^
```

This error could have been prevented by checking the CI: 
https://gepgitlab.laas.fr/gepetto/gepetto-viewer/pipelines/3113/builds

This error prevents the CI of hpp-doc to be pass: https://gepgitlab.laas.fr/gsaurel/hpp-doc/-/jobs/12087.

jmirabel, 2019-02-28:
Hum, I think I know the issue. I'll have a look.

nim65s, 2019-02-28:
Thanks :)

For now, on 16.04 / osg 3.2.1, I only found `ObjectCache` as a protected member of `osgDB::Registry`

jmirabel, 2019-02-28:
It was introduced somewhere between 3.2.1 and 3.4...

nim65s, 2019-02-28:
In 3.3.3: https://github.com/openscenegraph/OpenSceneGraph/commit/ed28ec97c708e31ea9030f2d88f3d0ace0e97baf#diff-b0eed203a64adbb82d6b2d33f3a5215c

closed 2019-02-28 by nim65s