Gepetto/gepetto-viewer #83: New release 4.3.1

jcarpent, 2019-03-05:
With the new fixes, I would need to make a new release of Gepetto-viewer. @jmirabel Is it fine for you?

nim65s, 2019-03-05:
No, please wait.
I have already a 4.4.0 ready. It is already in the binary repository, and I will finish to push it here by tomorrow.

jcarpent, 2019-03-05:
So, can you include my patch? which has been merged on devel by @jmirabel.

nim65s, 2019-03-05:
No, the patch will not be available on 4.4.0, it is too late. But we can make a 4.4.1 for apple users if you want. And we will not need to push this 4.4.1 on robotpkg.

jcarpent, 2019-03-05:
OK. Then, it will be fine if you can publish it now, for instance, I can take in charge the release 4.4.1.

jcarpent, 2019-03-05:
Just push the tag and on master, I will do the rest.

nim65s, 2019-03-05:
You don't need me, you can just make a 4.4.1 release. I pushed the 4.4.0 tag.

jcarpent, 2019-03-06:
4.4.1 is now out!

closed 2019-03-06 by jcarpent