Gepetto/gepetto-viewer #92: [Ubuntu 18.04] Movie recording broken (and GUI unreadable)

wxmerkt, 2019-05-10:
I just installed `gepetto-gui` from robotpkg and used the `Movie record` button - however, the status window looks like this:

[image]

On 18.04, `avconv` has become `ffmpeg` again - this may be the source of this issue?

Unrelated, the menu entries are all illegible until hovered

jmirabel, 2019-05-10:
If avconv command is not available, recording a movie won't work. However,
I am surprised you didn't get any warning or error message. You can go to
/tmp/gepetto-gui/ to get the images.

About the menu entries, I have no idea what's going on. Something to do
with Qt...

BTW, you can install doxygen-icon-theme to get proper icons for some
buttons.

nim65s, 2019-05-10:
gepetto-viewer from robotpkg does not yet comes with PythonQt.

@jmirabel : you told me that this would make some optional features unavailable. Maybe this is one of those ?

wxmerkt, 2019-05-10:
> If avconv command is not available, recording a movie won't work. However, I am surprised you didn't get any warning or error message. You can go to /tmp/gepetto-gui/ to get the images.

I aliased avconv=ffmpeg and it didn't work - there was neither an error message nor any output at all. I manually converted it to a movie later on.

> About the menu entries, I have no idea what's going on. Something to do with Qt...

FWIW, I only tried the Qt4 version, so it may vary on Qt5.

>  BTW, you can install doxygen-icon-theme to get proper icons for some buttons.

If this is required, shouldn't this be a dependency so that it auto-installs?

jmirabel, 2019-05-10:
> gepetto-viewer from robotpkg does not yet comes with PythonQt.

This does not use PythonQt.

jmirabel, 2019-05-10:
> If this is required, shouldn't this be a dependency so that it auto-installs?

It could. Probably, @nim65s didn't know it.

jmirabel, 2019-05-10:
> I aliased avconv=ffmpeg and it didn't work - there was neither an error message nor any output at all. I manually converted it to a movie later on.

A bash alias won't work. The command isn't run in a bash shell. Can you try to add a symlink in one of the directories of PATH ?

jmirabel, 2019-08-08:
Solved by #94 

> Unrelated, the menu entries are all illegible until hovered

You can create a Qt style sheet `stylesheet.txt`:
```
QMenu {
 background: white;
 color: black;
}
QMenu:selected {
 background: blue;
 color: white;
}
QToolTip {
 background: yellow;
 border: 1px solid black;
 color: black;
}
```
and run `gepetto-gui -stylesheet stylesheet.txt`

closed 2019-08-08 by jmirabel