Gepetto/gepetto-viewer #93: compile error s about osg

xuhuairuogu, 2019-05-24:
My environment is Ubuntu16.04 and I have installed openrave and openscenegraph3.2.1. The cmake can detect the openscenegraph successfully:
[image]
However, errors occur during the compiling like the following :
[image]

jmirabel, 2019-05-24:
The error message alone isn't enough. You must also provide the compilation command (last call to g++ before error, using `VERBOSE=1 make`).

xuhuairuogu, 2019-05-24:
[image]

jmirabel, 2019-05-24:
Do you have several versions of OpenSceneGraph installed ?

xuhuairuogu, 2019-05-24:
I search libosg and find there exists several versions.

jmirabel, 2019-05-24:
Then this is your issue. I think there is a missing `-L` option to the compilation command. Stackoverflow will be a better help for finding out how to solve your issue.

jmirabel, 2019-05-28:
I close because I believe this is not related to GV. Please, re-open if necessary.

closed 2019-05-28 by jmirabel