Gepetto/gepetto-viewer #97: [Ubuntu 18.04] gui items disappear behind transparent robot

stanislas-brossette, 2019-07-04:
Hello,
I'm using gepette-gui to visualize robot trajectories and physical quantities like CoM/CoP/DCM...
The issue I'm facing is that when displaying a transparent robot and a sphere representing the CoP, that sphere disappears behind the transparent robot, while the floor is clearly visible behind the robot.
Below is the core code that I use for displaying the scene.
Am I missing something or is that a bug/shortcoming?
 ```
from pinocchio.robot_wrapper import RobotWrapper
   
robot = RobotWrapper()
colorized_urdf = get_colorized_urdf(urdf_path, (1,1,1,0.7))
robot.initFromURDF(colorized_urdf, [root_path], pinocchio.JointModelFreeFlyer())
robot.initDisplay()
robot.viewer.gui.addFloor('hpp-gui/floor')
robot.loadDisplayModel('my_robot')
robot.display(utils.array_to_matrix(model_positions[:, 0]))
robot.viewer.gui.addSphere('current_CoP', 0.02, [1, 0, 0, 1])
robot.viewer.gui.addLine('current_CoP_line', [0.0, .0, 0.0], [0.0, 0.0, 0.3], [1, 0, 0, 1])
robot.viewer.gui.addToGroup('current_CoP', 'world')
robot.viewer.gui.addToGroup('current_CoP_line', 'world')

wait_for_start()

for i in range(len(times) - 1):
        robot.display(utils.array_to_matrix(model_positions[:, i]))
        dt = (times[i + 1] - times[i]) / speed
        if dt > 0.:
            time.sleep(dt)
        robot.viewer.gui.applyConfiguration('current_CoP', (cop[:, i].tolist() + [0.0, 0.0, 0.0, 1.0]))
        robot.viewer.gui.applyConfiguration('current_CoP_line', (cop[:, i].tolist() + [0.0, 0.0, 0.0, 1.0]))

    robot.display(utils.array_to_matrix(model_positions[:, -1]))
```

jmirabel, 2019-07-05:
Hi,

you hit two points:
- a known issue: Rendering transparent object uses a basic node ordering method. This is sufficient to ensure the order between transparent and non-transparent objects. However, undesired occlusion will happen between two transparent objects in a hard to predict way.

  Rendering transparent objects is not reliable because GV does not use a depth peeling algorithm. To my best knowledge, it is the only reliable algo for that purpose but it consumes a lot more computing power.

- a bug I was not aware of: the object rendering order is only modified by the setAlpha function and not by the setColor function...
  Can you try to add a `robot.viewer.gui.setFloatProperty("nodename", "Transparency", desiredAlpha)` ?

stanislas-brossette, 2019-07-05:
The setFloatProperty method fixed the problem, thank you :)

jmirabel, 2019-07-05:
In next version, "Transparency" will be replaced by "Alpha" to fix the incoherent behavior.

closed 2019-07-05 by jmirabel