Gepetto/gepetto-viewer #106: Default for avconv setting on 18.04 wrong

wxmerkt, 2019-10-07:
Relates to #92 and #94.

#94 added the ability to configure `avconv` to be `ffmpeg` or whatever we want. However, in the robotpkg distributed viewer, it still points to `avconv` on 18.04 by default - it is now `ffmpeg` again. Can this be fixed in the distributed version?

julesser, 2020-02-17:
This seems to be somehow related to an error I currently receive. When trying to record a movie of the crocoddyl example _bipedal_walking.py_ in the gepetto viewer, it fails with the following log:

```
Failed to start avconv. Either it is missing, or you may have insufficient permissions to invoke it. 

You can manually run avconv -r 25 -i /tmp/gepetto-gui/record/img_0_%d.jpeg -vf scale=trunc(iw/2)*2:trunc(ih/2)*2 -r 25 -vcodec libx264 /home/###hidden###/Videos/bipedalRecording.mp4
```

jmirabel, 2020-02-17:
@nim65s there are several possible workaround:
- use cmake to know what the default should be. Then, the line to change is https://github.com/Gepetto/gepetto-viewer/blob/0169d4b4da6c22fdc6354913b4a9fc1a8f768087/src/gui/settings.cc#L49
- otherwise, we could look for both and see if any is found.

nim65s, 2020-02-17:
Thanks for the heads-up, I had forgotten this.

jmirabel, 2020-02-25:
Fixed by #119

commit ef859216 (2020-02-17): [CMake] detect ffmpeg/avconv, fix #106 (#119)

* [CMake] detect ffmpeg/avconv, fix #106
commit 5f69640b (2020-02-17): [CMake] detect ffmpeg/avconv, fix #106 (#119)

* [CMake] detect ffmpeg/avconv, fix #106
closed 2020-02-25 by jmirabel