Gepetto/gepetto-viewer #112: robotpkg - Installation dependency conflicts

olimexsmart, 2019-12-09:
Hi, 

I am trying to install gepetto-viewer using robotpkg, which seems to be configured correctly because I am able to install and use other packages (e.g., pinocchio). However, when trying to install gepetto-viewer I get the following error

```
The following packages have unmet dependencies:
 robotpkg-gepetto-viewer : Depends: robotpkg-urdfdom-headers (= 0.3.0) but 1.0.4 is to be installed
                           Depends: robotpkg-urdfdom (= 0.3.0r2) but 1.0.3 is to be installed
E: Unable to correct problems, you have held broken packages.
olli@oberon:~/Documents/PhD/devel/projects/consim/script$ 
```

I am working on Linux Mint 19.2, which is based on Ubuntu 18.04. I have tried to clean my system by uninstalling most robotpkg packages, so that now I have only the following packages:
``` 
$ dpkg -l | grep robotpkg
ii  robotpkg-eigen3                            3.3.0                                           amd64        a C++ template library for linear algebra
ii  robotpkg-hpp-fcl+doc                       1.2.0                                           amd64        Humanoid Path Planner (collision detection and distance computations)
ii  robotpkg-libccd                            2.0                                             amd64        collision detection
ii  robotpkg-octomap                           1.9.0                                           amd64        An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-py36-eigenpy                      1.6.9                                           amd64        Eigen bindings to python
```
Can anybody help me figure out why it doesn' t work? Thanks in advance for your help.

jmirabel, 2019-12-09:
You probably have to install `robotpkg-py27-qt4-gepetto-viewer` and `robotpkg-py27-qt4-gepetto-viewer-corba`. Adapt it to your Python and Qt version.

olimexsmart, 2019-12-09:
Now my list of robotpkg packages looks like this but the problem persists

``` 
ii  robotpkg-eigen3                            3.3.0                                           amd64        a C++ template library for linear algebra
ii  robotpkg-example-robot-data                3.1.1                                           amd64        Set of robot URDFs for benchmarking and developed examples.
ii  robotpkg-hpp-fcl+doc                       1.2.0                                           amd64        Humanoid Path Planner (collision detection and distance computations)
ii  robotpkg-libccd                            2.0                                             amd64        collision detection
ii  robotpkg-octomap                           1.9.0                                           amd64        An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-omniorb                           4.2.3r1                                         amd64        ORB that implements the 2.6 specification of the CORBA
ii  robotpkg-pinocchio                         2.2.1                                           amd64        Efficient rigid body dynamics
ii  robotpkg-py36-eigenpy                      1.6.9                                           amd64        Eigen bindings to python
ii  robotpkg-py36-example-robot-data           3.1.1                                           amd64        Set of robot URDFs for benchmarking and developed examples.
ii  robotpkg-py36-omniorbpy                    4.2.3                                           amd64        Robust high-performance CORBA ORB for Python
ii  robotpkg-py36-pinocchio                    2.2.1                                           amd64        Efficient rigid body dynamics (python bindings)
ii  robotpkg-py36-qt4-gepetto-viewer           4.8.0                                           amd64        Humanoid Path Planner (Simple viewer based on OpenSceneGraph)
ii  robotpkg-py36-qt4-gepetto-viewer-corba     5.3.2                                           amd64        Humanoid Path Planner (Corba server for gepetto-viewer)
```

jmirabel, 2019-12-09:
What is the issue / error ?
Gepetto viewer seems correctly installed.

nim65s, 2019-12-09:
Hi @olimexsmart 

Where didi you find instructions about `robotpkg-gepetto-viewer` ? This is obsolete and should have done replaced py `robotpkg-py36-qt4-gepetto-viewer`, which you correctly have.

olimexsmart, 2019-12-10:
Hi, thanks for the quick response

Indeed now gepetto works fine

I just started a PhD with Andrea Del Prete and we were trying to set everything up. On his system `robotpkg-gepetto-viewer`, which is 16.04 based, worked fine. Don't know if the Ubuntu version matters in this question, but I would say it's the defining difference between our systems. BTW, he is going to substitute too his installation with these new gepetto packages.

Have a nice day,
Luca

closed 2019-12-10 by olimexsmart

Saeed-Mansouri, 2022-01-29:
I'm trying to install gepetto-viewer using robotpkg on Ubuntu 18.04. I have the same issue! How can I fix this error?

```
The following packages have unmet dependencies:
 robotpkg-gepetto-viewer : Depends: robotpkg-urdfdom-headers (= 0.3.0) but 1.0.4 is to be installed
                           Depends: robotpkg-urdfdom (= 0.3.0r2) but 1.0.3 is to be installed
E: Unable to correct problems, you have held broken packages.
```