Gepetto/gepetto-viewer #113: Installation on Ubuntu-18.04 fails

florent-lamiraux, 2019-12-10:
I have upgraded my computer to ubuntu-18.04 and the build of gepetto-viewer branch devel (just checked out) fails.

The problem seems to come from the fact that gepetto-gui is not linked to python libraries. As a consequence, at linking, I get several messages like
undefined reference to `PyString_Size' 

I solved the issue by adding a line in the src/CMakeLists.txt (commit f47465ae7f892e4fcc03dc8c1e1ad3f5d29276a2) but I am surprised that nobody got the same issue.

jmirabel, 2019-12-10:
Do you install GV with PythonQt ?

florent-lamiraux, 2019-12-10:
Yes, I use the default settings.

florent-lamiraux, 2019-12-10:
In fact, I realize that I configure with option -DPROJECT_USE_QT4=OFF although qt5 is not installed. I will try again and let you know.

florent-lamiraux, 2019-12-10:
Recompiling pythonqt with PythonQt_QT_VERSION=4, I get the same error.

jmirabel, 2019-12-10:
In PythonQt, there are two different branches for Qt4 and Qt5. Check that you are on the right branch.

florent-lamiraux, 2019-12-10:
pythonqt is on branch qt4, with `CMakeCache.txt` containing
`PythonQt_QT_VERSION:STRING=4`

florent-lamiraux, 2019-12-11:
The problem came from a mix up between python2 and python3. I solved it by adding option -DPYTHON_EXECUTABLE=/usr/bin/python3 at configuration in order to match PythonQt python version.

closed 2019-12-11 by florent-lamiraux