Gepetto/gepetto-viewer #125: Massive scaling issue and random joint orientation

guru-florida, 2020-03-02:
I can get the Talos robot to load fine. I've created my own urdf and srdf that loads in RViz2 fine but in Gepetto it's a jumble of links. It does load the links, but I have to zoom out as the model is about 40x or more larger than the talos model. In Blender the Talos STL and my STL/DAE models import as similar size so the models do not seem to be so different. Possibly the urdfdom is not respecting the STL units? (...but I dont really know what I'm talking about.)
[image]

The joint angles are also random. Possibly this will correct itself once the scale issue is resolved.

P.S. I looked on github, stackoverflow and looked for gepetto forum...is this the wrong place to post?

pFernbach, 2020-03-02:
Hello, I do not know if this is related to your issue, but as you mentionned the 'unit' in the DAE file I just tried and this tag seems to be ignored when loading visual meshes from DAE files. 
It is always using meter as unit. 

Also, did you correctly installed the 'osg-dae' package, with `apt-get install robotpkg-osg-dae`

guru-florida, 2020-03-02:
Thanks @pFernbach for taking the time to verify. Yes, I did install the osg-dae package, and furthermore I can see the DAEs load and appear. I can confirm in my DAE that the units are in mm.
`  <unit name="millimeter" meter="0.001"/>`
I remember setting this in Blender during STL to DAE conversion and it was recommended by the ROS2 URDF tutorial. Possibly I could convert to meters and see.

guru-florida, 2020-03-02:
Ok, using Blender again I exported all DAEs with meter units and the robot is more to size. Also the DAEs line up perfectly with my collision models (visibility ON) which are STL not DAEs. Model is looking good and connected except it is upside-down. lol

jmirabel, 2020-03-03:
#126 solves the DAE unit issue.

jmirabel, 2020-03-03:
> Model is looking good and connected except it is upside-down

This might be due to the up_axis tag of Collada files.

guru-florida, 2020-03-04:
I have my robot working and integrated in the bipedal_walk.py. My first successful movement made me LOL. You might get a laugh too :)
https://youtu.be/dqKk8PO9zCE

closed 2020-03-04 by guru-florida