Gepetto/gepetto-viewer #131: Plugin loading

nim65s, 2020-04-06:
Hi,

Running `gepetto-gui`, with gepetto-viewer & gepetto-viewer-corba raised the following errors:
```
"omniorbserver.so" :  "The shared library was not found."
Plugin "omniorbserver.so" not loaded: "The shared library was not found."
```

This .so is available at `/usr/lib/gepetto-gui-plugins`, so setting `QT_PLUGIN_PATH` to this path fixed the issue for me.

But depending on how I build GV & GV-corba, this is sometimes required and sometimes not.

Without this `QT_PLUGIN_PATH`, how is the plugin system able to find the plugins ?

jmirabel, 2020-04-06:
It looks for the plugin in *<path>/gepetto-gui-plugins* for each *path* in *LD_LIBRARY_PATH*.

nim65s, 2020-04-06:
Thanks, I understand better.

jmirabel, 2020-04-06:
I agree that the list of searched path is not exhaustive. Any suggestion is welcome.

nim65s, 2020-04-07:
I'd suggest to just add `${CMAKE_INSTALL_PREFIX}/lib/gepetto-gui-plugins`, and add a hint for `GEPETTO_GUI_PLUGIN_DIRS` in the error message.

closed 2020-04-22 by jmirabel