Gepetto/gepetto-viewer #141: Fail to compile gepetto-viewer in Ubuntu-16.04, qt4.

florent-lamiraux, 2020-05-13:
I try to recompile HPP with the latest releases to update the installation instruction.
When linking `gepetto-gui` I have many undefined symbols and I notice that the link command does not contain `-lpython`. Looking at the CMakeLists below,
https://github.com/Gepetto/gepetto-viewer/blob/f7b4a4497bf8276713a94ef0b768cc06111a97b2/src/CMakeLists.txt#L147
 I would expect something like
```
TARGET_LINK_LIBRARY(${LIBRARY_NAME} python)
```
I do not understand either why
```
${PROJECT_QT_LIBS}
${PYTHONQT_LIBRARIES})
```
is not inside the test
```
IF(GEPETTO_GUI_HAS_PYTHONQT)
```
The linking command line:
```
[ 91%] Linking CXX executable gepetto-gui
cd /home/florent/devel/test/src/gepetto-viewer/build-rel/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/gepetto-gui.dir/link.txt --verbose=1
/usr/lib/ccache/g++   -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG  -rdynamic  -losg -losgDB -losgFX -losgGA -losgParticle -losgSim -losgText -losgUtil -losgTerrain -losgManipulator -losgViewer -losgWidget -losgShadow -losgAnimation -losgVolume -lOpenThreads CMakeFiles/gepetto-gui.dir/gui/main.cc.o CMakeFiles/gepetto-gui.dir/gepetto-gui_autogen/mocs_compilation.cpp.o  -o gepetto-gui -Wl,-rpath,/home/florent/devel/test/src/gepetto-viewer/build-rel/src:/home/florent/devel/test/install/lib: -losg -losgDB -losgFX -losgGA -losgParticle -losgSim -losgText -losgUtil -losgTerrain -losgManipulator -losgViewer -losgWidget -losgShadow -losgAnimation -losgVolume -lOpenThreads libgepetto-viewer.so -losgQt -losg -losgDB -losgFX -losgGA -losgParticle -losgSim -losgText -losgUtil -losgTerrain -losgManipulator -losgViewer -losgWidget -losgShadow -losgAnimation -losgVolume -lOpenThreads -lboost_unit_test_framework -lboost_system -lSM -lICE -lX11 -lXext -lGL -lGLU /usr/lib/x86_64-linux-gnu/libQtOpenGL.so /usr/lib/x86_64-linux-gnu/libQtGui.so /usr/lib/x86_64-linux-gnu/libQtNetwork.so /usr/lib/x86_64-linux-gnu/libQtXml.so /usr/lib/x86_64-linux-gnu/libQtCore.so /home/florent/devel/test/install/lib/libPythonQt.so -lboost_python 
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyUnicodeUCS4_FromEncodedObject'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyNumber_InPlaceDivide'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyString_Size'
libgepetto-viewer.so: undefined reference to `PyString_FromStringAndSize'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyString_InternFromString'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyInt_AsLong'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyUnicodeUCS4_AsWideChar'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyString_AsString'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyString_FromFormat'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyClass_Type'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyString_FromString'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyFile_AsFile'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `Py_InitModule4_64'
libgepetto-viewer.so: undefined reference to `PyInt_Type'
libgepetto-viewer.so: undefined reference to `PyInt_FromLong'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyFile_FromString'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost_python.so: undefined reference to `PyNumber_Divide'
libgepetto-viewer.so: undefined reference to `PyString_Type'
collect2: error: ld returned 1 exit status
```

jmirabel, 2020-05-13:
> I do not understand either why
> 
> ```
> ${PROJECT_QT_LIBS}
> ${PYTHONQT_LIBRARIES})
> ```
> 
> is not inside the test
> 
> ```
> IF(GEPETTO_GUI_HAS_PYTHONQT)
> ```

There is only a weak reason: They should be empty when GEPETTO_GUI_HAS_PYTHONQT is FALSE and putting it here makes the code shorter.

For the rest, I have no idea. Maybe it's time to drop Qt4 support...

florent-lamiraux, 2020-05-13:
I will try to install with Qt5 and update the installation instructions accordingly.

nim65s, 2020-05-13:
I've been looking at this, and I found that adding `TARGET_LINK_LIBRARY(${LIBRARY_NAME} ${PYTHON_LIBRARIES})` after the ` TARGET_LINK_BOOST_PYTHON(${LIBRARY_NAME})` fixed the issue.

However, I wanted to dig further to try to understand why we didn't had this issue before. Does this mean that every time we link  to boost::python we also have to manually link to python ? And in all our other packages, we always linked to python every time we linked to boost::python just by chance ?

If so, shouldn't we just update the `TARGET_LINK_BOOST_PYTHON` macro to also link to python ?

Anyway, we can also just stop wondering, applying the fix, and go ahead.

For Qt4, I'm affraid it won't be possible to drop its support, while ubuntu is not providing osgqt with qt5.

Or if you are only talking about the HPP makefile, why not, but this mean that everybody will have to build openscenegraph, which is really boring.

jmirabel, 2020-05-13:
> I've been looking at this, and I found that adding `TARGET_LINK_LIBRARY(${LIBRARY_NAME} ${PYTHON_LIBRARIES})` after the ` TARGET_LINK_BOOST_PYTHON(${LIBRARY_NAME})` fixed the issue.

CMake should handle this automatically. I checked and the call to *target_link_libraries* **should** contain the *PUBLIC* keyword.

> For Qt4, I'm affraid it won't be possible to drop its support, while ubuntu is not providing osgqt with qt5.
> 
> Or if you are only talking about the HPP makefile, why not, but this mean that everybody will have to build openscenegraph, which is really boring.

Can't robotpkg also provide the Qt5 plugin where missing ?
The instructions to build that plugin are fairly easy. Something like:
```bash
apt source libopenscenegraph
untar and cd to source
mkdir build && cd build
cmake .. -DDESIRED_QT_VERSION=5 ..
cd osgQt
make install
```

This should only install the osgQt.so file, compatible with the repository version. The same applies to osg-dae, which should use the above procedure rather than the source repo I created long ago.

florent-lamiraux, 2020-05-13:
Is it possible to make a release 4.9.1 with the fix ? (a tag is enough.

nim65s, 2020-05-13:
Which `target_link_libraries` ? 

I don't think that is is easy to get sources from `apt source` in robotpkg. Let's ask Anthony: https://git.openrobots.org/issues/255

@florent-lamiraux : yes, I can easily make a 4.9.1 release.

nim65s, 2020-05-13:
Anyway, I think that with @jmirabel snippet, we can drop support for Qt4 at least in the HPP Makefile, can't we ?

jmirabel, 2020-05-13:
> Anyway, I think that with @jmirabel snippet, we can drop support for Qt4 at least in the HPP Makefile, can't we ?

could be. At the cost of having a more complex makefile.

nim65s, 2020-05-14:
Fixed by https://github.com/Gepetto/PythonQt/pull/6, thanks @florent-lamiraux !

closed 2020-05-14 by nim65s