Gepetto/gepetto-viewer #146: Migration from osgQt to osgQOpenGL

nim65s, 2020-06-12:
Hi,

In recent developments, the [osgQt project](https://github.com/openscenegraph/osgQt) removed everything related to osgQt (they just kept the name of the project) and switched to the new Qt5 preffered way and use QOpenGL:

https://github.com/openscenegraph/osgQt/pull/22

@jmirabel : could you evaluate the difficulty for us to move to the new interface ? Depending on the required work to make the switch, we can either do it, or I can finish my packaging with an older version of osgQt.

jmirabel, 2020-06-12:
It is hard to evaluate the time it takes to migrate.

Following your initial plan, you intend to provide a version of osgQt in robotpkg, that suits the need of gepetto-viewer ? This will not be different even if we switch so I think you should continue what you were doing.

nim65s, 2020-06-12:
Yes, I can continue on the initial plan, but we would need to either have some CMake magic that allows to use both osgQt or osgQOpenGL, or to synchronize well for the upgrade.