Gepetto/gepetto-viewer #149: Connection lost to the server

proyan, 2020-07-10:
Hi guys,
I'm following the devel branch in gepetto-viewer and gepetto-viewer-corba.

For the simple script below:
```python
from example_robot_data import loadTalos
r = loadTalos()
r.initViewer(loadModel=True)
```

I get the following error:
```
omniORB: From endpoint: giop:tcp:[::1]:12321. Detected GIOP 1.0 protocol error in input message. giopImpl10.cc:438. Connection is closed.
omniORB: To endpoint: giop:tcp:[::1]:12321. Send GIOP 1.0 MessageError because a protocol error has been detected. Connection is closed.
/local/rbudhira/devel/install/sot/lib/python2.7/dist-packages/pinocchio/robot_wrapper.py:238: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  self.viz.initViewer(*args, **kwargs)
```

I'm not sure why the connection gets closed. I've tried reinstalling gepetto-viewer, and I've tried restarting omni-nameserver by `sudo systemctl restart omniorb4-nameserver`

I have the following warning messages in the shell running `gepetto-gui`

```

Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
omniORB: From endpoint: giop:tcp:[::1]:46208. Detected GIOP 1.0 protocol error in input message. giopImpl10.cc:590. Connection is closed.
omniORB: To endpoint: giop:tcp:[::1]:46208. Send GIOP 1.0 MessageError because a protocol error has been detected. Connection is closed.
```

 I'm using python2.7, and Ubuntu16.04. Any help would be welcome,

Best
Rohan

proyan, 2020-07-10:
Surprisingly, when I try to connect to gepetto-gui in c++ using `https://github.com/jmirabel/pinocchio-gepetto-viewer`, I don't have any problem.

jmirabel, 2020-07-13:
From the *giop* address, it seems you use IPv6. I think I never tried to use IPv6 (at least, I never entered manually an IPv6 address). Although it should work out of the box, IPv4 is probably safer.

The nameserver isn't used anymore. You should be able to remove it.

C++ and Python access the server with the same way so it is weird that one works while the other does not. Do you have omniorb installed both from robotpkg and from the official repos ?

proyan, 2020-07-13:
I don't have robotpkg installations. I use either installation from source, or from apt.

proyan, 2020-07-13:
Is there an omni.logs or something like that that I can peruse?

proyan, 2020-07-13:
The error messages seem to be in https://fossies.org/linux/omniORB/src/lib/omniORB/orbcore/giopImpl10.cc. at Line 438 (for python client) and Line 590 (for gepetto-server)

jmirabel, 2020-07-13:
Are you sure that there cannot be a mix between your source installation and your apt installation of omniorb ?

proyan, 2020-07-13:
I don't have a source installation of omniorb

jmirabel, 2020-07-13:
Is it repeatable: it never works in Python but it always work in C++ ?

proyan, 2020-07-13:
Yes, I still have this error in python. And I don't have this error in c++

jmirabel, 2020-07-13:
I have no idea of what's going wrong. It seems unrelated to GV itself but rather a misuse of omniorb.

The error appears somewhere in function `initViewer` of *pinocchio/visualize/gepetto_visualizer.py*. Can you tell me what line exactly ?

proyan, 2020-07-13:
I found the issue. After you talked about parallel installtions, I found that I had gepetto-viewer python installation in both dist-packages and site-packages. I've removed `site-packages/gepetto` and now it works.
Thanks Joseph. I'll close this issue.

closed 2020-07-13 by proyan