Gepetto/gepetto-viewer #150: [question] Automated video recording

julesser, 2020-07-27:
Hi all, 
I was wondering whether there is a way to automatically record and save videos?

jmirabel, 2020-07-27:
What do you mean by *automatically* ? At the moment, you have `startCapture` and `stopCapture` for video generation, `captureFrame` for image generation and some extra for exporting motion to blender.

See section *Export of scenes to images or to blender* in https://gepettoweb.laas.fr/hpp/gepetto-viewer-corba/doxygen-html/interfacegepetto_1_1corbaserver_1_1GraphicalInterface.html for a list of available functions.

nim65s, 2020-07-27:
Here is another doc on blender export, to complete @jmirabel answer: https://github.com/Gepetto/gepetto-viewer-corba/blob/master/blender/README.md

julesser, 2020-07-27:
Thanks for the quick and helpful answers :)
> What do you mean by _automatically_ ?

_Automatically_ just in the sense of not having to press 'record' and 'save' for each simulation. 

So the fastest way here would be to call `startCapture` and `stopCapture` within the application for getting the image frames and then converting it manually to a video?

jmirabel, 2020-07-27:
> So the fastest way here would be to call `startCapture` and `stopCapture` within the application for getting the image frames and then converting it manually to a video?

Yes, this is the simplest way. `avconv` of `ffmpeg` is your friend for merging the images into a video.

closed 2020-07-27 by julesser