Gepetto/gepetto-viewer #154: object collision detection inside gepetto-viewer

ahmadgazar, 2020-10-05:
Hi,
I am new with gepetto-viewer. I am trying to visualize collisions of an object I placed inside the gui with the robot. what is the optimal way of doing so? Are there some tutorials/examples I can look into? Thanks in advance.

jmirabel, 2020-10-06:
gepetto-viewer doesn't do collision detection.

To emphasize that two objects are in collision, you can highlight them using `gui.setHighlight(name, highlight_id)`. The id goes from 0 to 8 (included). 0 means no highlight. 1 is best suited to highlight collision.

If you want to display the closest points, the best is to use `addLine`.

ahmadgazar, 2020-10-06:
Awesome ! that's what I needed. Thanks a lot.

closed 2020-10-06 by ahmadgazar