Gepetto/gepetto-viewer #163: Trouble building on 20.04

jakewelde, 2021-02-02:
Hi, I'm having trouble building GV, when I follow the build instructions and run `cmake ..` I get the following error, that I can't find `openscenegraph-osgQt5`. How do I install this dependency? I also wasn't able to find it manually myself. Many thanks!

```
-- Package version (ROS package.xml): 4.11.0
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
CMake Warning at cmake/doxygen.cmake:496 (MESSAGE):
  Failed to find Doxygen, documentation will not be generated.
Call Stack (most recent call first):
  cmake/post-project.cmake:58 (_SETUP_PROJECT_DOCUMENTATION)
  CMakeLists.txt:40 (PROJECT)

-- openscenegraph >= 3.2 is required.
-- Checking for module 'openscenegraph>=3.2'
--   Found openscenegraph, version 3.6.4
-- Pkg-config module openscenegraph v3.6.4 has been detected with success.
-- openthreads >= 2.6 is required.
-- Checking for module 'openthreads>=2.6'
--   Found openthreads, version 3.3.1
-- Pkg-config module openthreads v3.3.1 has been detected with success.
-- openscenegraph-osgQt5 is required.
-- Checking for module 'openscenegraph-osgQt5'
--   No package 'openscenegraph-osgQt5' found
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  cmake/pkg-config.cmake:336 (PKG_CHECK_MODULES)
  cmake/pkg-config.cmake:550 (ADD_DEPENDENCY)
  cmake-module/osgqt.cmake:11 (ADD_REQUIRED_DEPENDENCY)
  CMakeLists.txt:120 (ADD_OSGQT_DEPENDENCY)

-- Configuring incomplete, errors occurred!
See also "/home/jake/Projects/gepetto-viewer/_build/CMakeFiles/CMakeOutput.log".

```

florent-lamiraux, 2021-02-03:
You can find the required dependencies using robotpkg binary packages.
Add robotpkg repository by following the instruction at http://robotpkg.openrobots.org/debian.html.

Then
```
sudo apt update
sudo apt install robotpkg-qt5-qgv robotpkg-qt5-osgqt libpython3-all-dev python3-distutils libboost-python-dev
```
You should be able to compile after these instructions.

jmirabel, 2021-02-03:
As a complementary answer, if you install OSG from source, you also have to compile the https://github.com/openscenegraph/osgQt package. It has been separated from the main OSG package since version 3.5 or so.

jakewelde, 2021-02-03:
Thank you both for your quick and helpful responses!

Joseph, after posting that issue I tried to compile the package you mentioned but ran into many dependency/installation/package finding issues, so I just now tried what Florent suggested. I followed the instructions you shared and I was able to install these packages. I then ran `gepetto-gui` and then tried to use the visualizer from python as
```
from pinocchio.visualize import GepettoVisualizer

...

robot = RobotWrapper.BuildFromURDF(urdf_model_path, mesh_dir, pin.JointModelFreeFlyer())

robot.setVisualizer(GepettoVisualizer())
robot.initViewer()
robot.loadViewerModel("pinocchio")
robot.display(q0)

```

However, my python script reports errors like: 
```
/opt/ros/noetic/lib/python3/dist-packages/pinocchio/visualize/gepetto_visualizer.py:152: UserWarning: Error while loading geometry object: base_link_0
Error message:
No to_python (by-value) converter found for C++ type: boost::shared_ptr<pinocchio::fcl::FakeCollisionGeometry>
  self.loadViewerGeometryObject(visual,pin.GeometryType.VISUAL)
/opt/ros/noetic/lib/python3/dist-packages/pinocchio/visualize/gepetto_visualizer.py:152: UserWarning: Error while loading geometry object: left_leg_0
```

and my `gepetto-gui` reports errors like:

```
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
QOpenGLContext::swapBuffers() called with non-exposed window, behavior is undefined
Node "world/pinocchio/visuals/base_link_0" does not exist.
Node "world/pinocchio/visuals/left_leg_0" does not exist.
Node "world/pinocchio/visuals/left_base_0" does not exist.
Node "world/pinocchio/visuals/left_back_wheel_0" does not exist.
Node "world/pinocchio/visuals/left_front_wheel_0" does not exist.
Node "world/pinocchio/visuals/right_leg_0" does not exist.
Node "world/pinocchio/visuals/right_base_0" does not exist.
Node "world/pinocchio/visuals/right_back_wheel_0" does not exist.
Node "world/pinocchio/visuals/right_front_wheel_0" does not exist.
Node "world/pinocchio/visuals/gripper_pole_0" does not exist.
Node "world/pinocchio/visuals/left_gripper_0" does not exist.
Node "world/pinocchio/visuals/left_tip_0" does not exist.
Node "world/pinocchio/visuals/right_gripper_0" does not exist.
Node "world/pinocchio/visuals/right_tip_0" does not exist.
Node "world/pinocchio/visuals/head_0" does not exist.
Node "world/pinocchio/visuals/box_0" does not exist.
```

I saw another issue on here suggesting that `hpp-fcl ` dependency was missing and that was causing the by-value converter issues, but I've verified that it's installed via `sudo apt install robotpkg-hpp-fcl`. I see that I have some errors referencing my ROS installation of pinocchio - could that be the source of the problem? I've gone through several different routes trying to install pinocchio / gepetto viewer so I'm worried I may have wound up in a weird situation with regard to multiple installations / path issues.

Thanks again for helping me out!

jakewelde, 2021-02-03:
Update: I changed the order of my path so that now I'm using the `/opt/openrobots/` installation of pinocchio, and that seemed to resolve it. Thanks for your help!

closed 2021-02-03 by jakewelde