Gepetto/gepetto-viewer #164: No package 'openscenegraph' found

YunxiTang, 2021-02-10:
when i try to build gepetto-viewer, the error raised that 'No package 'openscenegraph' found' . but i actually already installed openscenegraph. As osgversion command works well on  my linux system.

jmirabel, 2021-02-10:
How did you install openscenegraph and which version did you install ?
openscenegraph is found with pkgconfig so you should have a valid PKG_CONFIG_PATH.

YunxiTang, 2021-02-12:
i use sudo install  to install openscenegraph.

florent-lamiraux, 2021-02-12:
Can you type `dpkg -l | grep openscenegraph` on your machine and copy paste the result here ?

jmirabel, 2021-02-12:
> i use sudo install to install openscenegraph.

I asked for the version as well. I think it is because you didn't install the development package of openscenegraph.

tlemarec, 2021-09-07:
Hello, same problem here.
I used `sudo apt install openscenegraph` and `dpkg -l | grep openscenegraph` returns :
```
ii  libopenscenegraph160:amd64                 3.6.4+dfsg1-3build2                         amd64        3D scene graph, shared libs
ii  openscenegraph                             3.6.4+dfsg1-3build2                         amd64        3D scene graph, utilities and examples (binaries)
```

nim65s, 2021-09-07:
This is missing the development package: can you try again after `sudo apt install libopenscenegraph-dev`  ?

tlemarec, 2021-09-07:
This worked thanks

closed 2021-09-07 by jmirabel