Gepetto/gepetto-viewer #167: Scene characteristics parametrization

vpadois, 2021-03-19:
Hello,
Is there a python example on how to parametrize the 3D scene appearance (ground, background,...) and load objects in the scene ? 
Best
Vincent

jmirabel, 2021-03-19:
You may have a look at https://github.com/gepetto/gepetto-viewer-corba/tree/devel/examples

They are quite basic though. What do you want to do ?

vpadois, 2021-03-19:
At this stage not much more than removing the background or adding frames associated to each bodies. Nothing very important but that could be handy.

jmirabel, 2021-03-19:
There is also the doxygen doc of the Python API. Available using standard Python help or https://gepettoweb.laas.fr/doc/gepetto/gepetto-viewer-corba/master/doxygen-html/interfacegepetto_1_1corbaserver_1_1GraphicalInterface.html

vpadois, 2021-03-19:
Thanks. I will explore the possibilities starting from there.
Best
Vincent

closed 2021-03-19 by vpadois