Gepetto/gepetto-viewer #168: Installing gepetto-viewer with conda

ntorresalberto, 2021-04-07:
Hi,
I've been trying out `pinocchio` but it doesn't seem to notice my `gepetto-viewer` install (https://anaconda.org/conda-forge/gepetto-viewer).
Both are installed with conda in the same environment.

I launch the viewer with `gepetto-gui` and then my pinocchio script  throws (the viewer examples throws the same error): 
```
$ python inverse-kinematics.py 
inverse-kinematics.py:53: UserWarning: Error while importing the viewer client.
Check whether gepetto-viewer is properly installed
  viz.initViewer()
inverse-kinematics.py:53: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  viz.initViewer()
Error while loading the viewer model. It seems you should start gepetto-viewer
'GepettoVisualizer' object has no attribute 'viewer'
```

jcarpent, 2021-04-07:
You should install `conda install gepetto-viewer-corba -c conda-forge` to get the Python bindings.

closed 2021-04-07 by jcarpent

ntorresalberto, 2021-04-07:
Indeed, that fixed it.
Could you please add it to the main readme?
I think something as simple as this would be of great help for those who want to give it a try without compiling:

## Conda Environment Setup
`conda install gepetto-viewer gepetto-viewer-corba -c conda-forge`

jcarpent, 2021-04-07:
> Indeed, that fixed it.
> Could you please add it to the main readme?
> I think something as simple as this would be of great help for those who want to give it a try without compiling:
> ## Conda Environment Setup
> 
> `conda install gepetto-viewer gepetto-viewer-corba -c conda-forge`

Could you open a PR with your suggestion?