Gepetto/gepetto-viewer #170: missing dependency: No package 'openscenegraph-osgQt5' found

ndehio, 2021-04-16:
Hi,
I tried to install gepetto-viewer from source (with a recent installation of openscenegraph 3.6) but the cmake fails. Which dependency am I missing? Could you please provide me a link?
thank you
Niels
```
ndehio@ndehio-5540:~/src/gepetto-viewer/build$ cmake ..
-- Package version (ROS package.xml): 4.11.0
-- CMake versions older than 3.12 may warn when looking to Boost components. Custom macros are used to find it.
-- openscenegraph >= 3.2 is required.
-- Checking for module 'openscenegraph>=3.2'
--   Found openscenegraph, version 3.6.5
-- Pkg-config module openscenegraph v3.6.5 has been detected with success.
-- openthreads >= 2.6 is required.
-- Checking for module 'openthreads>=2.6'
--   Found openthreads, version 3.3.1
-- Pkg-config module openthreads v3.3.1 has been detected with success.
-- openscenegraph-osgQt5 is required.
-- Checking for module 'openscenegraph-osgQt5'
--   No package 'openscenegraph-osgQt5' found
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  cmake/pkg-config.cmake:336 (PKG_CHECK_MODULES)
  cmake/pkg-config.cmake:550 (ADD_DEPENDENCY)
  cmake-module/osgqt.cmake:11 (ADD_REQUIRED_DEPENDENCY)
  CMakeLists.txt:120 (ADD_OSGQT_DEPENDENCY)
```

nim65s, 2021-04-16:
osgQt is available at https://github.com/openscenegraph/osgQt

ndehio, 2021-04-18:
Ok, I just installed osqQT from source (master branch)
```
[ 87%] Built target osgQOpenGL
[100%] Built target example_osgviewerQt
[ 87%] Built target osgQOpenGL
[100%] Built target example_osgviewerQt
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/lib64/pkgconfig/openscenegraph-osgQt.pc
Libraries were installed to /usr/locallib64.
You may need to update your ld.so configuration. 
-- Up-to-date: /usr/local/lib64/libosgQOpenGL.so.3.6.5
-- Up-to-date: /usr/local/lib64/libosgQOpenGL.so.145
-- Up-to-date: /usr/local/lib64/libosgQOpenGL.so
-- Up-to-date: /usr/local/include/osgQOpenGL/osgQOpenGLWidget
-- Up-to-date: /usr/local/include/osgQOpenGL/osgQOpenGLWindow
-- Up-to-date: /usr/local/include/osgQOpenGL/OSGRenderer
-- Up-to-date: /usr/local/include/osgQOpenGL/CullVisitorEx
-- Up-to-date: /usr/local/include/osgQOpenGL/GraphicsWindowEx
-- Up-to-date: /usr/local/include/osgQOpenGL/RenderStageEx
-- Up-to-date: /usr/local/include/osgQOpenGL/StateEx
-- Up-to-date: /usr/local/include/osgQOpenGL/Export
-- Up-to-date: /usr/local/share/OpenSceneGraph/bin/osgviewerQt
```
However, for installing gepetto-viewer the `cmake ..` is still failing with the same error as before

florent-lamiraux, 2021-04-19:
`gepetto-viewer` uses `pkg-config` to find where `osgQt` is installed. Make sure that `usr/local/lib64/pkgconfig` is in your `PKG_CONFIG_PATH`.

ndehio, 2021-04-19:
ok, I added `export PKG_CONFIG_PATH=usr/local/lib64/pkgconfig:$PKG_CONFIG_PATH` but the error is still the same.

jcarpent, 2021-04-19:
You have forgotten a `/` at the beginning of the path ;)

ndehio, 2021-04-19:
oh indeed, my mistake... But I still get 
```
--   No package 'openscenegraph-osgQt5' found
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  cmake/pkg-config.cmake:336 (PKG_CHECK_MODULES)
  cmake/pkg-config.cmake:550 (ADD_DEPENDENCY)
  cmake-module/osgqt.cmake:11 (ADD_REQUIRED_DEPENDENCY)
  CMakeLists.txt:120 (ADD_OSGQT_DEPENDENCY)
```

I noticed that OpenSceneGraph installs into `/usr/local/lib/` whereas osgQt installs into `/usr/local/lib64`. Is this a problem?

In my bashrc I have 
```
export PKG_CONFIG_PATH=/usr/local/lib64/pkgconfig:$PKG_CONFIG_PATH
export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig:$PKG_CONFIG_PATH
```

proyan, 2021-04-19:
I have the same issue.
For now, I replaced line 121
```
ADD_OSGQT_DEPENDENCY()
```
with 

```
ADD_REQUIRED_DEPENDENCY("openscenegraph-osgQt")
```
and it seems to work.

ndehio, 2021-04-19:
Thanks @proyan, that hack works for me, however now I have the following (probably related) problem at this line `PKG_CONFIG_USE_DEPENDENCY(${LIBRARY_NAME} ${OSGQT_DEPENDENCY})`
My cmake outputs this:
```
- Boost_PYTHON_LIBRARY: /usr/lib/x86_64-linux-gnu/libboost_python-py27.so
-- Skipping PythonQt settings
CMake Error at src/CMakeLists.txt:138 (PKG_CONFIG_USE_DEPENDENCY):
  PKG_CONFIG_USE_DEPENDENCY Macro invoked with incorrect arguments for macro
  named: PKG_CONFIG_USE_DEPENDENCY
```

proyan, 2021-04-19:
@ndehio 
Could you try to install the `topic/Qt4` branch of https://github.com/openscenegraph/osgQt instead of the `master` branch?
In my case, I didn't need to do any changes in `gepetto-viewer` package

ndehio, 2021-04-19:
Switching to Qt4 branch didn't help (but for some reason it still builds `libosgQt5`).

```
ndehio@ndehio-5540:~/src/osgQt/build$ git status
On branch Qt4
Your branch is up to date with 'origin/topic/Qt4'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        ./

nothing added to commit but untracked files present (use "git add" to track)
ndehio@ndehio-5540:~/src/osgQt/build$ sudo make install
[100%] Built target osgQt5
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/lib64/pkgconfig/openscenegraph-osgQt5.pc
Libraries were installed to /usr/locallib64.
You may need to update your ld.so configuration. 
-- Up-to-date: /usr/local/lib64/libosgQt5.so.3.6.5
-- Up-to-date: /usr/local/lib64/libosgQt5.so.145
-- Up-to-date: /usr/local/lib64/libosgQt5.so
-- Up-to-date: /usr/local/include/osgQt/Export
-- Up-to-date: /usr/local/include/osgQt/GraphicsWindowQt
-- Up-to-date: /usr/local/include/osgQt/QFontImplementation
-- Up-to-date: /usr/local/include/osgQt/QGraphicsViewAdapter
-- Up-to-date: /usr/local/include/osgQt/QWidgetImage
-- Up-to-date: /usr/local/include/osgQt/QWebViewImage
```
and
```
ndehio@ndehio-5540:~/src/gepetto-viewer/build$ cmake ..
-- Package version (ROS package.xml): 4.11.0
-- CMake versions older than 3.12 may warn when looking to Boost components. Custom macros are used to find it.
-- openscenegraph >= 3.2 is required.
-- Checking for module 'openscenegraph>=3.2'
--   Found openscenegraph, version 3.6.5
-- Pkg-config module openscenegraph v3.6.5 has been detected with success.
-- openthreads >= 2.6 is required.
-- Checking for module 'openthreads>=2.6'
--   Found openthreads, version 3.3.1
-- Pkg-config module openthreads v3.3.1 has been detected with success.
-- openscenegraph-osgQt is required.
-- Checking for module 'openscenegraph-osgQt'
--   No package 'openscenegraph-osgQt' found
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  cmake/pkg-config.cmake:336 (PKG_CHECK_MODULES)
  cmake/pkg-config.cmake:550 (ADD_DEPENDENCY)
  CMakeLists.txt:121 (ADD_REQUIRED_DEPENDENCY)

-- Configuring incomplete, errors occurred!
See also "/home/ndehio/src/gepetto-viewer/build/CMakeFiles/CMakeOutput.log".
```

jcarpent, 2021-04-19:
You may rather use:
```
conda install gepetto-viewer-corba -c conda-forge
```

nim65s, 2021-04-19:
@ndehio : On which distribution are you trying to build this ?

ndehio, 2021-04-19:
@nim65s It is a Ubuntu 18.04

nim65s, 2021-04-19:
Thanks. I'll try to reproduce this in a clean environment.

nim65s, 2021-04-19:
It looks like everything is working for me with this Dockerfile, assuming Python 2.7 & Qt 5:

```dockerfile
FROM ubuntu:18.04

RUN apt-get update -qqy && DEBIAN_FRONTEND=noninteractive apt-get install -qqy \
    cmake \
    doxygen \
    gcc \
    git \
    libboost-all-dev \
    libomniorb4-dev \
    libpythonqt-qtall-qt5-python2-dev \
    libqt5svg5-dev \
    libqt5xmlpatterns5-dev \
    omniidl \
    omniidl-python \
    omniorb \
    omniorb-idl \
    omniorb-nameserver \
    pkg-config \
    pyqt5-dev \
    python2.7-dev \
    python-omniorb \
    qtbase5-private-dev \
    qtmultimedia5-dev \
    openscenegraph-3.4 \
    libopenscenegraph-3.4-dev

WORKDIR /src

RUN git clone https://github.com/openscenegraph/osgQt.git \
 && mkdir osgQt/build \
 && cd osgQt/build \
 && git checkout topic/Qt4 \
 && cmake .. \
 && make -sj4 \
 && make install

RUN git clone --recursive https://github.com/gepetto/gepetto-viewer.git \
 && mkdir gepetto-viewer/build \
 && cd gepetto-viewer/build \
 && git checkout devel \
 && cmake .. \
 && make -sj4 \
 && make install

RUN git clone --recursive https://github.com/gepetto/gepetto-viewer-corba.git \
 && mkdir gepetto-viewer-corba/build \
 && cd gepetto-viewer-corba/build \
 && git checkout devel \
 && cmake .. \
 && make -sj4 \
 && make install

CMD make -C gepetto-viewer/build test \
 && make -C gepetto-viewer-corba/build test
 ```
 
 There are multiple things included in this Dockerfile that were not mentioned in the previous answers to this issue.
 
 Is this helping ?

ndehio, 2021-04-20:
Tomorrow I will have time to investigate further.
Note that I installed OpenSceneGraph version 3.6 (not 3.4) from source.

jmirabel, 2021-04-21:
It seems you guys have mixed a few things. Macro `ADD_OSGQT_DEPENDENCY` should obviously not be shortcut. It does the logic of looking for the good package.

For OSG >= 3.5.5, gepetto-viewer will look for openscenegraph-osgQt for Qt4 and openscenegraph-osgQt5.

@ndehio start from scratch again and do
- choose your version of Qt,
- install osgQt for your version of Qt,
- compile gepetto-viewer with the correct Qt version (set CMake variable `PROJECT_USE_QT4` to `ON` if you want Qt4, which I don't recommend).

ndehio, 2021-04-22:
I tried several combinations with OpenSceneGraph version 3.6 and 3.4 installed from source which all didn't workout.

Finally, I did `sudo apt-get install openscenegraph-3.4 libopenscenegraph-3.4-dev` and then I was able to install osgQt, gepetto-viewer and gepetto-viewer-corba without further problems. Thanks!

closed 2021-04-22 by ndehio

Shervina-lim, 2022-01-04:
> It looks like everything is working for me with this Dockerfile, assuming Python 2.7 & Qt 5:
> 
> ```dockerfile
> FROM ubuntu:18.04
> 
> RUN apt-get update -qqy && DEBIAN_FRONTEND=noninteractive apt-get install -qqy \
>     cmake \
>     doxygen \
>     gcc \
>     git \
>     libboost-all-dev \
>     libomniorb4-dev \
>     libpythonqt-qtall-qt5-python2-dev \
>     libqt5svg5-dev \
>     libqt5xmlpatterns5-dev \
>     omniidl \
>     omniidl-python \
>     omniorb \
>     omniorb-idl \
>     omniorb-nameserver \
>     pkg-config \
>     pyqt5-dev \
>     python2.7-dev \
>     python-omniorb \
>     qtbase5-private-dev \
>     qtmultimedia5-dev \
>     openscenegraph-3.4 \
>     libopenscenegraph-3.4-dev
> 
> WORKDIR /src
> 
> RUN git clone -b topic/Qt4 https://github.com/openscenegraph/osgQt.git \
>  && mkdir osgQt/build \
>  && cd osgQt/build \
>  && cmake .. \
>  && make -sj4 \
>  && make install
> 
> RUN git clone --recursive -b devel https://github.com/gepetto/gepetto-viewer.git \
>  && mkdir gepetto-viewer/build \
>  && cd gepetto-viewer/build \
>  && cmake .. \
>  && make -sj4 \
>  && make install
> 
> RUN git clone --recursive -b devel https://github.com/gepetto/gepetto-viewer-corba.git \
>  && mkdir gepetto-viewer-corba/build \
>  && cd gepetto-viewer-corba/build \
>  && cmake .. \
>  && make -sj4 \
>  && make install
> 
> CMD make -C gepetto-viewer/build test \
>  && make -C gepetto-viewer-corba/build test
> ```
> 
> There are multiple things included in this Dockerfile that were not mentioned in the previous answers to this issue.
> 
> Is this helping ?

Followed ur steps. Successfully installed gepetto-viewer. Thanks a lot!!