Gepetto/gepetto-viewer #174: Citing Gepetto Viewer

gabrielebndn, 2021-05-26:
Hi guys,
at Wandercraft we are using Gepetto Viewer for visualizing the robot movements. We are mentioning this in a paper and we would like to properly cite your work. However, I could not think of any good reference, nor I could find one by searching online.

As far as I remember there isn't any, but I still prefer to double-check with you: what's the most appropriate way to cite Gepetto Viewer? Should we just use a generic citation to the website?

@jmirabel, what do you recommend?

jmirabel, 2021-05-27:
Hi,
There is no article associated to this project so using a link to github is
the best. Thank you for doing this.

gabrielebndn, 2021-05-28:
Ok, thanks for your answer!

closed 2021-06-02 by jmirabel