Gepetto/gepetto-viewer #175: gepetto-gui: "scale" text box being filled with garbage values

proyan, 2021-06-10:
Hi guys,

When setting the radius of a cylinder through the gepetto-gui API, I noticed some weird behaviour. When I press "del" or "backspace" to delete the existing values, the tail end of the value keeps filling up with what I think is garbage. I attach a video to show what I observed

https://user-images.githubusercontent.com/12423606/121509098-971f3d00-c9e6-11eb-8776-5046d2373a30.mp4

I am currently on the commit 11d71fb6bde478e0c4d079fec0cf8cf5cf173bf9 in the devel branch.

florent-lamiraux, 2021-06-14:
Strange behavior. It seems to me that the bug is more related to Qt than to gepetto-gui since gepetto-gui simply uses textboxes. Does it happen for other widgets ?

jmirabel, 2021-06-14:
The *garbage* comes from the spin box. When you enter a value, the value is converted to float and send to the cylinder node. The cylinder node updates the geometry and update the display to the float value.

Are you able to enter the radius you want ?

proyan, 2021-06-14:
The same value filling occurs for the other boxes as well. I was able to replicate it for height and alpha text boxes.
>Are you able to enter the radius you want ?

I finally did it through the python interface. It was a bit tricky to do it through the gui

jmirabel, 2021-06-14:
I agree that it does seem very easy.

I'm surprised though because I did use it. I wouldn't have been satisfied with what you experienced so I would have modified the code. You use Qt5 ?

proyan, 2021-06-14:
> I'm surprised though because I did use it. I wouldn't have been satisfied with what you experienced so I would have modified the code. You use Qt5 ?

Yes, I'm using Qt5