Gepetto/gepetto-viewer #176: CMake Error at CMakeLists.txt:31 (INCLUDE)

IoannisDadiotis, 2021-06-25:
Hi,

I am trying to setup in my Ubuntu 18.04. When `cmake ..` in my build folder, I am getting the following error:
```
CMake Error at CMakeLists.txt:31 (INCLUDE):
  INCLUDE could not find load file:

    cmake/base.cmake

CMake Error at CMakeLists.txt:32 (INCLUDE):
  INCLUDE could not find load file:

    cmake/boost.cmake

CMake Error at CMakeLists.txt:33 (INCLUDE):
  INCLUDE could not find load file:

    cmake/ide.cmake

CMake Error at CMakeLists.txt:34 (INCLUDE):
  INCLUDE could not find load file:

    cmake/apple.cmake

CMake Error at CMakeLists.txt:35 (INCLUDE):
  INCLUDE could not find load file:

    cmake/python.cmake

CMake Error at CMakeLists.txt:36 (INCLUDE):
  INCLUDE could not find load file:

    cmake/cxx-standard.cmake

CMake Error at CMakeLists.txt:40 (COMPUTE_PROJECT_ARGS):
  Unknown CMake command "COMPUTE_PROJECT_ARGS".

-- Configuring incomplete, errors occurred!
```
I checked the folder `/my_workspace/src/gepetto-viewer/cmake` and it is empty.

Thank you.

olivier-stasse, 2021-06-25:
Dear @IoannisDadiotis, thank your interest.
You did not use the `--recursive` option when cloning your repository.
It is highly recommended to install Gepetto-Viewer through a binary package.

IoannisDadiotis, 2021-06-26:
Thank you @olivier-stasse.

After some additional issues, I finally built and installed without errors.

However, when typping `gepetto-gui` on a terminal I get:
 ```gepetto-gui: error while loading shared libraries: libgepetto-viewer.so: cannot open shared object file: No such file or directory```

The file `/usr/local/lib/libgepetto-viewer.so` exists.

I read that gepetto-viewer used `pkgconfig` to locate compiled files. I am not sure if this makes sense but I tried to locate the `libgepetto-viewer.so` by adding to my `.bashrc` the following:
```
export PKG_CONFIG_PATH=/usr/local/lib64/pkgconfig:$PKG_CONFIG_PATH
export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig:$PKG_CONFIG_PATH
```
I also include ```export GEPETTO_GUI_PLUGIN_DIRS=${CMAKE_INSTALL_PREFIX}/lib``` as suggested.

nim65s, 2021-06-26:
Hi @IoannisDadiotis ,

pkg-config is used at configure or compile time, not at runtime.
To allow your executable to find its shared objects at runtime, you can use `export LD_LIBRARY_PATH=/usr/local/lib`

IoannisDadiotis, 2021-06-26:
Hi @nim65s, I can now launch the gui. Notice that in the terminal I launch it, I get 
```
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
```

Morover, while trying to use `crocoddyl` and doing `python -m crocoddyl.examples.quadrupedal_gaits "plot" "display"` (without the display option I have no issues) I get:
```
*** SOLVE walking ***
/opt/openrobots/lib/python2.7/site-packages/pinocchio/robot_wrapper.py:272: UserWarning: Error while importing the viewer client.
Check whether gepetto-gui is properly installed
  self.viz.initViewer(*args, **kwargs)
/opt/openrobots/lib/python2.7/site-packages/pinocchio/robot_wrapper.py:272: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  self.viz.initViewer(*args, **kwargs)
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/opt/openrobots/lib/python2.7/site-packages/crocoddyl/examples/quadrupedal_gaits.py", line 102, in <module>
    display = crocoddyl.GepettoDisplay(anymal, 4, 4, cameraTF, frameNames=[lfFoot, rfFoot, lhFoot, rhFoot])
  File "/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.py", line 100, in __init__
    self._addRobot()
  File "/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.py", line 239, in _addRobot
    self.robot.loadViewerModel(rootNodeName="robot")
  File "/opt/openrobots/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 294, in loadViewerModel
    self.viz.loadViewerModel(*args, **kwargs)
  File "/opt/openrobots/lib/python2.7/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 137, in loadViewerModel
    gui = self.viewer.gui
AttributeError: 'GepettoVisualizer' object has no attribute 'viewer'
```

florent-lamiraux, 2021-06-28:
The error seems to come from this block: https://github.com/stack-of-tasks/pinocchio/blob/7e3f96e59047b40e678a53b3877b401af4b6d0bd/bindings/python/pinocchio/visualize/gepetto_visualizer.py#L37.
Can you try to run the commands one by one to check where it fails ?

IoannisDadiotis, 2021-06-28:
@florent-lamiraux, I am not sure if this is the optimal way but I did it by adding breakpoints.

It seems to fail in  [this line](https://github.com/stack-of-tasks/pinocchio/blob/7e3f96e59047b40e678a53b3877b401af4b6d0bd/bindings/python/pinocchio/visualize/gepetto_visualizer.py#L38) while the `UserWarning: Error while importing the viewer client.
Check whether gepetto-gui is properly installed
  self.viz.initViewer(*args, **kwargs)` is raised from [previous line](https://github.com/stack-of-tasks/pinocchio/blob/7e3f96e59047b40e678a53b3877b401af4b6d0bd/bindings/python/pinocchio/visualize/gepetto_visualizer.py#L35).

IoannisDadiotis, 2021-06-28:
I think I have missed `gepetto-viewer-corba` which is important for the display option.

When `sudo apt update && sudo apt install robotpkg-py27-qt4-gepetto-viewer-corba`
I get:
```
Hit:1 http://it.archive.ubuntu.com/ubuntu bionic InRelease
Hit:2 http://it.archive.ubuntu.com/ubuntu bionic-updates InRelease     
Hit:3 http://it.archive.ubuntu.com/ubuntu bionic-backports InRelease   
Hit:4 http://dl.google.com/linux/chrome/deb stable InRelease           
Hit:5 http://archive.ubuntu.com/ubuntu bionic InRelease                
Hit:6 https://desktop-download.mendeley.com/download/apt stable InRelease
Hit:7 http://packages.microsoft.com/repos/code stable InRelease        
Ign:8 http://robotpkg.openrobots.org/packages/debian/pub bionic InRelease
Get:9 http://security.ubuntu.com/ubuntu bionic-security InRelease [88,7 kB]
Hit:10 http://robotpkg.openrobots.org/packages/debian/pub bionic Release
Hit:12 http://packages.ros.org/ros/ubuntu bionic InRelease             
Get:13 http://security.ubuntu.com/ubuntu bionic-security/main amd64 DEP-11 Metadata [48,5 kB]
Get:14 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 DEP-11 Metadata [60,6 kB]
Get:15 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 DEP-11 Metadata [2.464 B]
Fetched 200 kB in 1s (149 kB/s)     
Reading package lists... Done
Building dependency tree       
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 robotpkg-py27-qt4-gepetto-viewer-corba : Depends: robotpkg-qt4-osgqt (= 3.5.7r1) but 3.5.7r3 is to be installed or
                                                   robotpkg-qt4-osgqt (= 3.5.7r2) but 3.5.7r3 is to be installed
                                          Depends: robotpkg-py27-qt4-gepetto-viewer (= 4.9.0r2) or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.6.0) or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.9.0r1) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.9.0r2) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.8.0) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.9.0r3) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.9.0r3) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.9.0) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.7.0) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.8.0) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.10.1r1) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.9.0) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.7.0) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.9.0r1) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.10.1r1) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.10.1r2) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.10.1r2)
E: Unable to correct problems, you have held broken packages.
```

florent-lamiraux, 2021-06-28:
It seems to fail to start the CORBA client. Can you run gepetto-gui with option `-v` and copy paste the terminal output here ?

IoannisDadiotis, 2021-06-28:
```
Configuration:
	Configuration file:              	settings
	Log file:                        	standard output
	Predefined robots:               	robots
	Predefined environments:         	environments
	User configuration search path:  	/home/ioannis/.config/gepetto-gui
	System configuration search path:	/usr/local/etc/gepetto-gui

Options:
	Verbose:                          	1
	No plugin:                        	0
	Use omni name service:            	0
	Refresh rate:                     	30

Screen capture options:
	Directory:                        	/home/ioannis/Pictures/gepetto-gui
	Filename:                         	screenshot
	Extension:                        	png
	Avconv command:                   	/usr/bin/ffmpeg
	Avconv input options:             	-r 25
	Avconv output options:            	-vf scale=trunc(iw/2)*2:trunc(ih/2)*2 -r 25 -vcodec libx264

omniORB configuration
	-ORBendPoint = :::12321
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
```

florent-lamiraux, 2021-06-28:
I did not see your previous message. I need actually to install robotpkg-gepetto-viewer-corba.
Maybe you  can try `sudo apt upgrade` and then retry `sudo apt install robotpkg-py27-qt4-gepetto-viewer-corba`.

IoannisDadiotis, 2021-06-28:
I get the same errors:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 robotpkg-py27-qt4-gepetto-viewer-corba : Depends: robotpkg-qt4-osgqt (= 3.5.7r1) but 3.5.7r3 is to be installed or
                                                   robotpkg-qt4-osgqt (= 3.5.7r2) but 3.5.7r3 is to be installed
                                          Depends: robotpkg-py27-qt4-gepetto-viewer (= 4.9.0r2) or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.6.0) or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.9.0r1) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.9.0r2) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.8.0) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.9.0r3) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.9.0r3) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.9.0) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.7.0) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.8.0) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.10.1r1) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.9.0) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.7.0) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.9.0r1) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.10.1r1) or
                                                   robotpkg-py27-qt4-gepetto-viewer+doc (= 4.10.1r2) but it is not going to be installed or
                                                   robotpkg-py27-qt4-gepetto-viewer (= 4.10.1r2)
E: Unable to correct problems, you have held broken packages.
```

florent-lamiraux, 2021-06-28:
It seems that the binary packages are being rebuilt: http://robotpkg.openrobots.org/rbulk/robotpkg/index.html?.
Wait until it is done and try again.

jmirabel, 2021-06-29:
You cannot install gepetto viewer from source and gepetto-viewer corba from binaries...

IoannisDadiotis, 2021-06-29:
@jmirabel Thanks for the clarification. I am trying to install gepetto-viewer-corba from source. When `make install` I get:
```
/usr/bin/cmake -H/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba -B/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build/CMakeFiles /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build'
make -f src/CMakeFiles/generate_idl_cpp.dir/build.make src/CMakeFiles/generate_idl_cpp.dir/depend
make[2]: Entering directory '/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build'
cd /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/src /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build/src /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build/src/CMakeFiles/generate_idl_cpp.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build'
make -f src/CMakeFiles/generate_idl_cpp.dir/build.make src/CMakeFiles/generate_idl_cpp.dir/build
make[2]: Entering directory '/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build'
make[2]: Nothing to be done for 'src/CMakeFiles/generate_idl_cpp.dir/build'.
make[2]: Leaving directory '/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build'
[  9%] Built target generate_idl_cpp
make -f src/CMakeFiles/generate_idl_python.dir/build.make src/CMakeFiles/generate_idl_python.dir/depend
make[2]: Entering directory '/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build'
cd /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/src /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build/src /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build/src/CMakeFiles/generate_idl_python.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build'
make -f src/CMakeFiles/generate_idl_python.dir/build.make src/CMakeFiles/generate_idl_python.dir/build
make[2]: Entering directory '/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build'
[  9%] Generating Python stubs for graphical-interface
cd /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build/src && /usr/bin/omniidl -bpython -I/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/idl -Cgepetto/viewer/corba -C/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build/src -K -Wbpackage=gepetto.corbaserver /home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/idl/gepetto/viewer/graphical-interface.idl
omniidl: Could not import back-end 'python'
omniidl: Maybe you need to use the -p option?
omniidl: (The error was 'No module named 'omniidl_be.python'')
src/CMakeFiles/generate_idl_python.dir/build.make:63: recipe for target 'src/gepetto/viewer/corba/gepetto/viewer/corba/graphical-interface_idl.py' failed
make[2]: *** [src/gepetto/viewer/corba/gepetto/viewer/corba/graphical-interface_idl.py] Error 1
make[2]: Leaving directory '/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build'
CMakeFiles/Makefile2:1472: recipe for target 'src/CMakeFiles/generate_idl_python.dir/all' failed
make[1]: *** [src/CMakeFiles/generate_idl_python.dir/all] Error 2
make[1]: Leaving directory '/home/ioannis/fromlaas_ws/src/gepetto-viewer-corba/_build'
Makefile:143: recipe for target 'all' failed
make: *** [all] Error 2
```
To reach this point it was needed to `sudo apt-get install libomniorb4-dev omniidl`

jmirabel, 2021-06-29:
Why don't you go for binaries ?

IoannisDadiotis, 2021-06-29:
> You cannot install gepetto viewer from source and gepetto-viewer corba from binaries...

I have already installed gepetto-viewer from source. Based on your comment I cannot install gv-corba from binaries unless I re-install gepetto-viewer from binaries. Do you think this is a better option?

jmirabel, 2021-06-29:
Remove everything you have installed and use the binaries.

IoannisDadiotis, 2021-06-29:
So I will first need to re-install `gepetto-viewer` and `osgQt` from binaries. Could you guide me through this? I only found [documentation](https://github.com/Gepetto/gepetto-viewer#setup) for installing from source.

nim65s, 2021-06-29:
You don't need osgQt. If you run `sudo apt install robotpkg-py3\*-qt5-gepetto-viewer-corba`, everything you need will automatically come

IoannisDadiotis, 2021-06-29:
Can I adjust to python2.7 and qt4 (since I have  `Qt version 4.8.7` and default `Python 2.7.17`)? Something like `sudo apt install robotpkg-py27\*-qt4-gepetto-viewer-corba` ?

nim65s, 2021-06-29:
I suggest `robotpkg-py27-qt5-gepetto-viewer-corba`. This will automatically setup the Qt5 required dependency on your system, without interfering with your existing Qt 4.8.7. Using the viewer in Qt4 is becoming deprecated, and you might not have the latest version, features and bugfixes.

IoannisDadiotis, 2021-06-29:
Thank you. Well, I played around with the examples and they seem to run as expected through terminal.

I then tried to run an [example](https://github.com/loco-3d/crocoddyl/blob/master/examples/notebooks/arm_manipulation.ipynb) from the notebooks in a virtual environment with python2.7 interpreter in Pycharm (while running `gepetto-gui` in a terminal) and I got:
```
/opt/openrobots/lib/python2.7/site-packages/pinocchio/robot_wrapper.py:272: UserWarning: Error while importing the viewer client.
Check whether gepetto-gui is properly installed
  self.viz.initViewer(*args, **kwargs)
/opt/openrobots/lib/python2.7/site-packages/pinocchio/robot_wrapper.py:272: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  self.viz.initViewer(*args, **kwargs)
Traceback (most recent call last):
  File "/home/ioannis/PycharmProjects/test_crocoddyl/arm_manipulation.py", line 14, in <module>
    display = crocoddyl.GepettoDisplay(robot, cameraTF=cameraTF, floor=False)
  File "/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.py", line 100, in __init__
    self._addRobot()
  File "/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.py", line 239, in _addRobot
    self.robot.loadViewerModel(rootNodeName="robot")
  File "/opt/openrobots/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 294, in loadViewerModel
    self.viz.loadViewerModel(*args, **kwargs)
  File "/opt/openrobots/lib/python2.7/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 137, in loadViewerModel
    gui = self.viewer.gui
AttributeError: 'GepettoVisualizer' object has no attribute 'viewer'
``` 
As far as I can understand, it looks like I should add some paths to the interpreter but I am not sure which one. I tried adding the following:
```
/opt/openrobots/lib
/usr/local/lib
```
as in the former there are `libgepetto-viewer-corba.so`  `libgepetto-viewer.so` files while in the latter `libgepetto-viewer.so` file, but that didn't solve.

IoannisDadiotis, 2021-06-30:
Solved by augmenting interpreter's path with `/usr/lib/python2.7/dist-packages`. 

Copied the code from [unicycle_utils](https://github.com/loco-3d/crocoddyl/blob/master/examples/notebooks/unicycle_utils.py) to my project as a workaround

Thanks to everyone for the support.

closed 2021-06-30 by IoannisDadiotis