Gepetto/gepetto-viewer #181: Meshes are never deleted

florent-lamiraux, 2022-01-27:
To display an octree, I create an obj file that I then load through CORBA by calling method `WindowsManager::addMesh`.
This function calls `LeafNodeCollada::create` that calls `LeafNodeCollada::init`.

After this line,
https://github.com/Gepetto/gepetto-viewer/blob/7e420b7d07013f97198235000e943592f69d9c1a/src/leaf-node-collada.cpp#L110
the number of `referenceCount` of `collada_ptr_` is 2 although I expect it to be 1.
As a consequence, in the desctructor the test
https://github.com/Gepetto/gepetto-viewer/blob/7e420b7d07013f97198235000e943592f69d9c1a/src/leaf-node-collada.cpp#L439
is always false and the object is never removed from the cache.
As a consequence, reloading the object after deleting the node displayes the previous object eventhough the files has been modified.

I tried to add a method to explicitely remove the object from the cache when deleting the node, but it does not solve the problem.
I also tried to decrement manually the counter by adding
```
collada_ptr_->unref()
```
after line 110 above, but I got a segmentation fault.

Does anyone know whether it is normal that the reference count is equal to 2 just after creating the object ?

jmirabel, 2022-01-31:
I would say yes. One reference hold by the cache, one hold by `collada_ptr_`.

florent-lamiraux, 2022-01-31:
You are right. The problem is that the object is not removed from the cache here:
https://github.com/florent-lamiraux/gepetto-viewer/blob/ec82cb0c80082c4c4c13195705aa2ccae3439216/src/leaf-node-collada.cpp#L459
I am investigating.

jmirabel, 2022-01-31:
I think we have to store the option used to load the mesh and pass it to `removeFromObjectCache`.

jmirabel, 2022-01-31:
https://github.com/openscenegraph/OpenSceneGraph/blob/34a1d8bc9bba5c415c4ff590b3ea5229fa876ba8/src/osgDB/ObjectCache.cpp#L79

florent-lamiraux, 2022-01-31:
Looking at OpenSceneGraph code you are probably right. I will try.

florent-lamiraux, 2022-03-01:
Solved by https://github.com/Gepetto/gepetto-viewer/pull/183.

closed 2022-03-01 by florent-lamiraux