Gepetto/gepetto-viewer #185: compilation error

AlessioMosca, 2022-02-17:
I can not figure out why I am not able to compile the package. I got the following error:

error: ‘void QAbstractButton::clicked(bool)’ is protected within this context
     QObject::connect(button, &QAbstractButton::clicked, [this]() { get(); });
                                                ^~~~~~~
In file included from /usr/include/qt4/QtGui/qcheckbox.h:45:0,
                 from /usr/include/qt4/QtGui/QCheckBox:1,
                 from /home/amosca/gepetto-viewer/src/node-property.cpp:21:
/usr/include/qt4/QtGui/qabstractbutton.h:127:10: note: declared protected here
     void clicked(bool checked = false);
          ^~~~~~~
/home/amosca/gepetto-viewer/src/node-property.cpp:158:76: error: no matching function for call to ‘gepetto::viewer::VoidProperty::connect(QPushButton*&, void (QAbstractButton::*)(bool), gepetto::viewer::VoidProperty::guiEditor()::<lambda()>)’
     QObject::connect(button, &QAbstractButton::clicked, [this]() { get(); });
                                                                            ^
In file included from /usr/include/qt4/QtCore/QObject:1:0,
                 from /home/amosca/gepetto-viewer/include/gepetto/viewer/node-property.h:29,
                 from /home/amosca/gepetto-viewer/src/node-property.cpp:17:
/usr/include/qt4/QtCore/qobject.h:204:17: note: candidate: static bool QObject::connect(const QObject*, const char*, const QObject*, const char*, Qt::ConnectionType)
     static bool connect(const QObject *sender, const char *signal,
                 ^~~~~~~
/usr/include/qt4/QtCore/qobject.h:204:17: note:   candidate expects 5 arguments, 3 provided
/usr/include/qt4/QtCore/qobject.h:217:17: note: candidate: static bool QObject::connect(const QObject*, const QMetaMethod&, const QObject*, const QMetaMethod&, Qt::ConnectionType)
     static bool connect(const QObject *sender, const QMetaMethod &signal,
                 ^~~~~~~
/usr/include/qt4/QtCore/qobject.h:217:17: note:   candidate expects 5 arguments, 3 provided
/usr/include/qt4/QtCore/qobject.h:337:13: note: candidate: bool QObject::connect(const QObject*, const char*, const char*, Qt::ConnectionType) const
 inline bool QObject::connect(const QObject *asender, const char *asignal,
             ^~~~~~~
/usr/include/qt4/QtCore/qobject.h:337:13: note:   no known conversion for argument 2 from ‘void (QAbstractButton::*)(bool)’ to ‘const char*’
src/CMakeFiles/gepetto-viewer.dir/build.make:611: recipe for target 'src/CMakeFiles/gepetto-viewer.dir/node-property.cpp.o' failed
make[2]: *** [src/CMakeFiles/gepetto-viewer.dir/node-property.cpp.o] Error 1
make[2]: uscita dalla directory "/home/amosca/gepetto-viewer/build"
CMakeFiles/Makefile2:1471: recipe for target 'src/CMakeFiles/gepetto-viewer.dir/all' failed
make[1]: *** [src/CMakeFiles/gepetto-viewer.dir/all] Error 2
make[1]: uscita dalla directory "/home/amosca/gepetto-viewer/build"
Makefile:143: recipe for target 'all' failed
make: *** [all] Error 2

The output of the cmake is the following:

-- Package version (ROS package.xml): 4.12.1
-- CMake versions older than 3.12 may warn when looking to Boost components. Custom macros are used to find it.
-- The CXX compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Performing Test R-pedantic
-- Performing Test R-pedantic - Success
-- Performing Test R-Wno-long-long
-- Performing Test R-Wno-long-long - Success
-- Performing Test R-Wall
-- Performing Test R-Wall - Success
-- Performing Test R-Wextra
-- Performing Test R-Wextra - Success
-- Performing Test R-Wcast-align
-- Performing Test R-Wcast-align - Success
-- Performing Test R-Wcast-qual
-- Performing Test R-Wcast-qual - Success
-- Performing Test R-Wformat
-- Performing Test R-Wformat - Success
-- Performing Test R-Wwrite-strings
-- Performing Test R-Wwrite-strings - Success
-- Performing Test R-Wconversion
-- Performing Test R-Wconversion - Success
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
CMake Warning at cmake/doxygen.cmake:496 (MESSAGE):
[... 78 lines cut ...]
    "/home/amosca/gepetto-viewer/build/src/__/include/gepetto/gui/dialog/moc_pluginmanagerdialog.cxx"
    "/home/amosca/gepetto-viewer/build/src/__/include/gepetto/gui/moc_ledindicator.cxx"
    "/home/amosca/gepetto-viewer/build/src/__/include/gepetto/gui/moc_mainwindow.cxx"
    "/home/amosca/gepetto-viewer/build/src/__/include/gepetto/gui/moc_node-action.cxx"
    "/home/amosca/gepetto-viewer/build/src/__/include/gepetto/gui/moc_osgwidget.cxx"
    "/home/amosca/gepetto-viewer/build/src/__/include/gepetto/gui/moc_selection-event.cxx"
    "/home/amosca/gepetto-viewer/build/src/__/include/gepetto/gui/moc_selection-handler.cxx"
    "/home/amosca/gepetto-viewer/build/src/__/include/gepetto/gui/moc_shortcut-factory.cxx"
    "/home/amosca/gepetto-viewer/build/src/__/include/gepetto/gui/moc_tree-item.cxx"
    "/home/amosca/gepetto-viewer/build/src/__/include/gepetto/gui/moc_windows-manager.cxx"
    "/home/amosca/gepetto-viewer/build/src/qrc_images.cxx"

  from processing by AUTOMOC and AUTOUIC.  If any of the files should be
  processed, set CMP0071 to NEW.  If any of the files should not be
  processed, explicitly exclude them by setting the source file property
  SKIP_AUTOGEN:

    set_property(SOURCE file.h PROPERTY SKIP_AUTOGEN ON)

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/amosca/gepetto-viewer/build

what am i doing wrong?

nim65s, 2022-02-17:
Hi, I'm not sure how the Qt4 support is still working or not. Could you try with Qt5 to see if this help ?

jmirabel, 2022-02-17:
Your osg version is rather old. You must first check what Qt version this OSG version supports. If it supports Qt5, make sure it was compiled with Qt5. Then, force usage of Qt5 by switching CMake option `PROJECT_USE_QT4` to `FALSE`. For old OSG version, this variable is true by default.

jmirabel, 2022-02-17:
Have a look at the files in `cmake-modules` directory to see the logic. The files are no so numerous and they are short.

AlessioMosca, 2022-06-13:
Hi guysp, many thanks for your help. @jmirabel  I  followed your suggestion and it worked. Many thanks.