Gepetto/gepetto-viewer #186: Robotpkg packaging

jmirabel, 2022-03-16:
Hi @nim65s,

Could you update gepetto-viewer binaries on robotpkg:
- [ ] to include QCustomPlot (`cmake -DBUILD_PY_QCUSTOM_PLOT=ON` should be sufficient) and
- [ ] to add `oxygen-icon-theme` as an apt dependency ?

For the record, QCustomPlot license is GPL and it is included in gepetto-viewer as a plugin. It does not change the license of gepetto-viewer. It implies that Python plugins based on QCustomPlot are GPL.

jmirabel, 2022-09-08:
Hi @nim65s 
Would you have time to do this ?

nim65s, 2022-09-08:
First part is trivial and will be handled soon. For the second, I am not sure yet, but I'll check with the first :)

nim65s, 2022-11-16:
I think this is OK now. @jmirabel : could you confirm and close this ?

jmirabel, 2022-11-19:
I guess it's fine. I haven't had the chance to test yet. I'll re-open if there is an issue.

Thank you.

closed 2022-11-19 by jmirabel