Gepetto/gepetto-viewer #193: build fails on ubuntu 18.04 clean install

TJBetter, 2022-07-01:
Hello,

Thank you all for making this repo public. I have been using it on ubuntu 20.04 with no problem. When I tried to build it on ubuntu 18.4 following https://github.com/Gepetto/gepetto-viewer/issues/170, I cannot compile gepeto-viewer. The error message with make -sj4 gives:
[image]

Any thoughts? Thank you.

nim65s, 2022-07-02:
Hi,

Thanks for your report.

Could you describe how you installed the dependencies, so I can try to reproduce your issue ?

Also, if you prefer something that works out of the box, you can use binary packages. We provide debs for ubuntu 18.04 with robotpkg: http://robotpkg.openrobots.org/debian.html ; so in your case with python 2 and qt5, you after setting up the source, you should be able to `sudo apt install robotpkg-py27-qt5-gepetto-viewer-corba`.

Conda is also a solution, but I'm no expert, and it's not compatible with other python stuff from your system.

TJBetter, 2022-07-11:
Thank you for your attention. I have decided to move to ubuntu 20 and now I solved this problem. Sorry I had to remove the screenshot since it reveals sensitive information.

closed 2022-07-11 by TJBetter