Gepetto/gepetto-viewer #196: adding stl mesh makes other meshes disappear

andreadelprete, 2022-07-27:
Hi, I am trying to visualize a UR5 robot together with a table in gepetto viewer. I first initialize the viewer and load the robot model using a pinocchio RobotWrapper, as usual. Then I load the table stl model using `addMesh("world/table", "path-to-stl-file")`. The result is that as soon as I load the table model, the table appears and the robot disappears from the viewer. If I do the operations in the opposite order (first loading the table, then the robot model) I get the opposite result, namely the table disappears as the robot appears. Any idea what might be going on?

I am working on a Ubuntu 20.04 virtual machine, using the ur5 model of example-robot-data and the following versions of gepetto-viewer packages:
```
ii  robotpkg-py38-qt5-gepetto-viewer+doc       4.12.0r2                            amd64        Humanoid Path Planner (Simple viewer based on OpenSceneGraph)
ii  robotpkg-py38-qt5-gepetto-viewer-corba     5.6.0                               amd64        Humanoid Path Planner (Corba server for gepetto-viewer)
```

florent-lamiraux, 2022-08-01:
Do you have a script to reproduce the error ?

andreadelprete, 2022-08-01:
This code can reproduce the error:
```
from example_robot_data.robots_loader import load

STL_FILE = '/home/adelprete/devel/src/locosim/ros_impedance_controller/worlds/models/tavolo/mesh/tavolo.stl'

robot = load('ur5')
robot.initViewer(loadModel=True)
robot.display(robot.q0)

import time 
time.sleep(3.0)

robot.viz.viewer.gui.addMesh('world/table', STL_FILE)
```
The stl file can be found [here](https://github.com/iit-DLSLab/ros_impedance_controller/blob/ad4128f45bbf2f5b17dd3e9e465b3c8d66250042/worlds/models/tavolo/mesh/tavolo.stl).

florent-lamiraux, 2022-08-01:
I found the UR5
[image]

florent-lamiraux, 2022-08-01:
Do you have another one ?
I love this game: "find the UR5".

florent-lamiraux, 2022-08-01:
More seriously. I think there is a problem of scale. Your stl is expressed in millimeters.

andreadelprete, 2022-08-01:
Thanks @florent-lamiraux ! Sorry for the silly problem! :D

closed 2022-08-01 by andreadelprete