Gepetto/gepetto-viewer #205: QOpenGLContext::swapBuffers() called with non-exposed window, behavior is undefined

ZhengmaoHe, 2022-12-06:
Hello, 
I successfully installed gepetto-viewer and its dependencies on ubuntu20.04, but I can't make it display robots normally. When I run the sample program, the terminal running gepetto-gui will output
````qt5ct: using qt5ct plugin
libpng warning: iCCP: known incorrect sRGB profile
qt5ct: D-Bus global menu: no
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
QOpenGLContext::swapBuffers() called with non-exposed window, behavior is undefined
Node "c_base_link_0" does not exist.
Node "c_shoulder_link_0" does not exist.
Node "c_upper_arm_link_0" does not exist.
Node "c_forearm_link_0" does not exist.
Node "c_wrist_1_link_0" does not exist.
Node "c_wrist_2_link_0" does not exist.
Node "c_wrist_3_link_0" does not exist.
Node "c_ee_link_0" does not exist.
Node "v_base_link_0" does not exist.
Node "v_shoulder_link_0" does not exist.
Node "v_upper_arm_link_0" does not exist.
Node "v_forearm_link_0" does not exist.
Node "v_wrist_1_link_0" does not exist.
Node "v_wrist_2_link_0" does not exist.
Node "v_wrist_3_link_0" does not exist.
````
At the same time, GEPETTO GUI window displays only one coordinate axis.
My sample program is as follows:
````
#define PINOCCHIO_MODEL_DIR "/home/mao/Desktop/Third-Party/pinocchio/models"

int main(int argc, char ** argv)
{
  using namespace pinocchio;
 
  const std::string model_path = PINOCCHIO_MODEL_DIR + std::string("/example-robot-data/robots");
  const std::string mesh_dir = PINOCCHIO_MODEL_DIR;
  const std::string urdf_filename = model_path + "/ur_description/urdf/ur5_robot.urdf";
 
  // Load the urdf model
  Model model;
  pinocchio::urdf::buildModel(urdf_filename,model);
  GeometryModel collision_model;
  pinocchio::urdf::buildGeom(model, urdf_filename, COLLISION, collision_model, mesh_dir);
  GeometryModel visual_model;
  pinocchio::urdf::buildGeom(model, urdf_filename, VISUAL, visual_model, mesh_dir);
  std::cout << "model name: " << model.name << std::endl;
 
    //! [Create a viewer]
    pinocchio::gepetto::Viewer viewer (model, &visual_model, &collision_model);
    viewer.initViewer("pinocchio");
    viewer.loadViewerModel("ur5");

    viewer.addFrame(model.nframes-1);
    //! [Create a viewer]

    //! [Display a configuration]
    viewer.display(pinocchio::neutral(model));
    //! [Display a configuration]
    viewer.display(pinocchio::randomConfiguration(model));
}
````
This problem has been bothering me for several days. I hope you can give me some tips. Thanks!

jmirabel, 2022-12-07:
It probably unrelated to the `swapBuffers` warning.

The `Node ... does not exist` makes me think the issue is in `viewer.loadViewerModel`. I guess the objects are created but not added to the scene.

> Also, is there another visualizers that I can use to easily display the robot while programming in c++?

Not that I know of.

florent-lamiraux, 2022-12-07:
Hello @MariaBana 
Difficult to reproduce you issue. There are no include in your sample C++ code.
Can you embed your code in a minimal package with a CMakeLists.txt so that we can compile it ?
Did you install from binary packages or from source. If so which versions of pinocchio, gepetto-viewer and gepetto-viewer-corba do you use ?

Otherwise can you look in the Body Tree what are the object present ?

MariaBana, 2022-12-07:
Hi @florent-lamiraux 
Thanks to your generous help, 
>  Can you embed your code in a minimal package with a CMakeLists.txt so that we can compile it ? 

I have written a more concise code here(simply replacing the urdf path of https://github.com/stack-of-tasks/pinocchio-gepetto-viewer/blob/master/tests/ur5.cc)
[learn-pinocchio.zip](https://github.com/Gepetto/gepetto-viewer/files/10174773/learn-pinocchio.zip)
>  Did you install from binary packages or from source. If so which versions of pinocchio, gepetto-viewer and gepetto-viewer-corba do you use ?

They are installed from source. The pinocchio I use is the pinocchio3-preview branch (because I need to use the closed chain manipulator), gepetto-viewer and gepetto-viewer-corba are the master branches.
Other than that, I installed libopenscenegraph-dev3.6.4 and libopenscenegraph160 3.6.4 from binary packages, and osg3.6.4 from source.
>  Otherwise can you look in the Body Tree what are the objects present ?

It seems that it does not load the body tree correctly, here is a screenshot of the window.
[image]

MariaBana, 2022-12-07:
@jmirabel 
> I guess the objects are created but not added to the scene.

Thanks very much for the tip, could you give me more guidance to verify this guess? These lines of code are written like the official example, so it should be fine, right?
```
    pinocchio::gepetto::Viewer viewer (model, &visual_model, &collision_model);
    viewer.initViewer("pinocchio");
    viewer.loadViewerModel("ur5");
    viewer.addFrame(model.nframes-1);
    viewer.display(pinocchio::neutral(model));
```

florent-lamiraux, 2022-12-07:
I am not so familiar with the bindings between pinocchio and gepetto-viewer. I cannot find which package installs
`#include <pinocchio/gepetto/viewer.hpp>`

florent-lamiraux, 2022-12-07:
It seems to work on my computer
[image]

florent-lamiraux, 2022-12-07:
Maybe you can try with another urdf model.

jmirabel, 2022-12-07:
Yes, the issue is that the models are not loaded. I suspect that the meshes cannot be found. Check your `ROS_PACKAGE_PATH`.

MariaBana, 2022-12-07:
@jmirabel 
I can't use pinocchio through ROS because I need pinocchio3-preview to use closed chain dynamics.
But I installed pinocchio2 to compare. I changed the name of the base.dae in the meshes file and the following error is output in pinocchio2
```
mao@mao:~/Desktop/myProject/learn-pinocchio/build$ ./learn-pinocchio 
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Could not load resource /home/mao/Desktop/Third-Party/pinocchio/models/example-robot-data/robots/ur_description/meshes/ur5/visual/base.dae
Unable to open file "/home/mao/Desktop/Third-Party/pinocchio/models/example-robot-data/robots/ur_description/meshes/ur5/visual/base.dae".
Hint: the mesh directory may be wrong.
Aborted (core dumped)
```
This means it can find the meshes file. The same learn-pinocchio program in pinocchio3, on the other hand, runs normally, so I think there may be a problem with the read meshes step in pinocchio3?

However, neither pinocchio2 nor 3 can display the robot properly, their gepetto-gui reports the same error, as does the body tree.

MariaBana, 2022-12-07:
@florent-lamiraux 
I tried several models provided in pinocchio and they all report the same error. And compared with pinocchio2 (see the my last reply for details), I think the problem may not be loading the model?

jmirabel, 2022-12-07:
I'm pretty sure the issue is unrelated. Either the paslth is wrong or the file got corrupted somehow. Check whether you can open the file with meshlab. Double check the path.

MariaBana, 2022-12-08:
@jmirabel 
I made sure the path and files are OK, I have now installed the osg-dae plugin and successfully displayed the robot in pinocchio2, but the **same program** doesn't work in pinocchio3, still reporting the same error.

florent-lamiraux, 2022-12-08:
Can you open an issue in pinocchio explaining that displaying models using pinocchio 2 + pinocchio-gepetto-viewer works but that pinocchio 3 + gepetto-viewer does not, refer to this issue and ask for what has changed ?

closed 2022-12-09 by jmirabel