Gepetto/gepetto-viewer #225: installation steps to run pinocchio in ubuntu20

RamCharanThota, 2024-06-25:
# Pinocchio and Gepetto-viewer Setup Guide for Ubuntu 20.04

This guide provides steps to successfully install Pinocchio and Gepetto-viewer on Ubuntu 20.04 with Python 3.8 and Qt5.

## Step 1: Install Pinocchio

Follow the installation steps provided on the Pinocchio website:

- Visit [Pinocchio Installation Guide](https://stack-of-tasks.github.io/pinocchio/download.html) and follow the instructions.

    - Don't forget to Configure environment variables as specified in the guide.

## Step 2: run the following commands

$ sudo apt update
$ sudo apt install libopenscenegraph-dev qt5-default
$ sudo apt install -qqy robotpkg-py*-qt5-gepetto-viewer
$ sudo apt install -qqy robotpkg-py*-qt5-gepetto-viewer-corba
$ sudo apt install -qqy robotpkg-py*-example-robot-data

## Step 3 : add the following to lines to your bashrc file 

 (need to check whether these lines are really necessary)
export GEPETTO_GUI_PLUGIN_DIRS=${CMAKE_INSTALL_PREFIX}/lib
export PKG_CONFIG_PATH=/usr/local/lib64/pkgconfig:$PKG_CONFIG_PATH

nim65s, 2024-06-26:
I guess `robotpkg-py*-qt5-gepetto-viewer-corba` should automatically come with `libopenscenegraph-dev qt5-default` and `robotpkg-py*-qt5-gepetto-viewer`, so there should be no need to install those explicitly

RamCharanThota, 2024-06-26:
is it?. Let me double check about "robotpkg-py*-qt5-gepetto-viewer-corba". I have spent lot of time by doing trail and error to install gepetto-viewer properly. so i might have missed something.