Gepetto/gepetto-viewer #226: Does gepetto-viewer depend on the ffmpeg openscenegraph plugin?

traversaro, 2024-07-02:
Hello gepetto-viewer mantainers, thanks a lot for the work on this repo! 

I wanted to understand in which way `gepetto-viewer` uses ffmpeg. From what I understand from the code (but I  may be completely wrong), it seems that it requires a `ffmpeg` command to be present at runtime (see https://github.com/Gepetto/gepetto-viewer/blob/5261b98f88673ce6ff9dd4fd1a03c557d58006b0/src/gui/settings.cc#L480), but it does not require the `ffmpeg` openscenegraph plugin to be present? 

To give you a bit of context, the question emerges from the conda-forge packaging of gepetto-viewer, in https://github.com/conda-forge/openscenegraph-feedstock/issues/29#issuecomment-2194897729 . In conda-forge, we are trying to get rid of ffmpeg 4.4, however openscenegraph was never updated to run with a newer ffmpeg (see upstream issue in https://github.com/openscenegraph/OpenSceneGraph/issues/1111, basically all the effort is now on new projects).

Ideally, we would like to disable the compilation of the ffmpeg plugin of openscengraph, exactly how Debian/Ubuntu packagers did for the openscenegraph apt package in Ubuntu 24.04 (https://git.launchpad.net/ubuntu/+source/openscenegraph/commit/?h=ubuntu/noble&id=a453c15270e96863f6c174c2ad1f6bf08e917111). 

However, in the past it seemed that there was a runtime dependency of gepetto-viewer on the ffmpeg plugin of openscengraph (see @ymontmarin comment in https://github.com/conda-forge/openscenegraph-feedstock/issues/29#issuecomment-1757847330). However, as now the ffmpeg plugin of osg is not present in Ubuntu 24.04, I wanted to double check this. If indeed  `gepetto-viewer` requires the ffmpeg plugin of osg, how are you dealing with this on Ubuntu 24.04? Are you building your own openscenegraph and ffmpeg package? 

Thanks a lot in advance for any reply or info!

traversaro, 2024-07-02:
By the way I noticed totally by chance that @nim65s today opened a PR to add gepetto-viewer to nixpkgs (https://github.com/NixOS/nixpkgs/pull/324018), but I do not know enough about nix to understand if that is using osg's ffmpeg plugin or not. : )

nim65s, 2024-07-03:
Hi @traversaro,

ffmpeg is optional, and used only at runtime, to capture the screen.

In CMake, we have `find_program(FFMPEG ffmpeg)` because on some systems, this program might be called `avconv` instead ; but clearly this is an optional dependency, and it should be fine if neither ffmpeg or avconv are available at configure or runtime.
https://github.com/Gepetto/gepetto-viewer/blob/5261b98f88673ce6ff9dd4fd1a03c557d58006b0/CMakeLists.txt#L190-L202

I was not aware of the existence of an ffmpeg plugin for openscenegraph, and I don't think we need that to record a video: we use a direct `QProcess("ffmpeg", "something something")` call.

I did not add this ffmpeg dependency in nixpkgs yet. I'll do that only when an user will ask for the feature ;)

It is also possible to record an entire screen with OBS for a quick video, or to use another feature of gepetto-viewer which is exporting a sequence to blender and have a cleaner rendering

traversaro, 2024-07-03:
Great, thanks! Indeed to actually do a end to end test yesterday I installed gepetto-viewer, I manually deleted the osg ffmpeg plugin, and the video was still created, so I am quite certainty that everything in gepetto-viewer can work without any osg ffmpeg, thanks!

closed 2024-07-03 by traversaro